aredn/aredn #1992: Add Connection type Icon on Local connections consistent with the way it is shown on Remote Nodes

k1ky, 2025-03-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Not a problem - new feature request to show connection information Icon to the right of the Babel Icon for Local Node connections consistent with how it is shown for Remote Nodes. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Display connection Icons as described above in the Local Node section.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
Like this:

[image]

aanon4, 2025-03-15:
Local nodes are *always* DtD connected, so the icon would always be the same and no extra information would be conveyed. Is that useful?

k1ky, 2025-03-16:
For some of my setups at home, EOC operation settings, and relief operations - local nodes are a combination of DTD and RF connections.

aanon4, 2025-03-16:
A local node is defined as being connected by a DtD link and no further away than 100 meters.

closed 2025-03-30 by aanon4

k1ky, 2025-05-06:
I still contend that the behavior of the connection status type should be reflected consistent in the Local Node area as there can be RF connections less than 100 meters as stated above in Home and EOC Situations.