aredn/aredn #1993: WAN Port Forwarding is applied to mesh requests

VA2XJM, 2025-03-16:
### Describe the bug
Port forwarding is applied to requests made from inside the mesh.

- [X] Does this issue still occur on the latest version? (**Required**) -> **Nightly: 20250315-a7033633**

### Expected behavior
Port forwarding should only occurs from WAN side. So mesh internal requests on port 80 or 8080 should not be forwarded.

### Support data
Can be provided on request, but this can be replicated easily.

### Additional context
We have some devices that we forward ports 80 or 8080. When trying to reach the admin GUI through the mesh, we get redirected toward the hosted service instead of the admin GUI. It has been introduced lately because the GUI was accessible through mesh before.

When locally connected to the affected nodes, we can reach the GUI as usual.

aanon4, 2025-03-17:
Completely unrelated, but your logs are full of this:
```
[143938.101558] nf_conntrack: nf_conntrack: table full, dropping packet
[144004.283229] nf_conntrack: nf_conntrack: table full, dropping packet
[144005.822132] nf_conntrack: nf_conntrack: table full, dropping packet
[144053.822362] nf_conntrack: nf_conntrack: table full, dropping packet
[144078.297125] nf_conntrack: nf_conntrack: table full, dropping packet
[144192.017453] nf_conntrack: nf_conntrack: table full, dropping packet
[144249.139276] nf_conntrack: nf_conntrack: table full, dropping packet
[144249.140410] nf_conntrack: nf_conntrack: table full, dropping packet
[144249.145199] nf_conntrack: nf_conntrack: table full, dropping packet
[144250.409413] nf_conntrack: nf_conntrack: table full, dropping packet
[144446.960833] nf_conntrack: nf_conntrack: table full, dropping packet
[144721.022263] nf_conntrack: nf_conntrack: table full, dropping packet
[144723.803001] nf_conntrack: nf_conntrack: table full, dropping packet
[144847.299570] nf_conntrack: nf_conntrack: table full, dropping packet
[144847.301027] nf_conntrack: nf_conntrack: table full, dropping packet
[144847.314638] nf_conntrack: nf_conntrack: table full, dropping packet
[144847.316504] nf_conntrack: nf_conntrack: table full, dropping packet
```
Last time I saw this there was a virus on the network using the WAN pathway to run a brute force ssh attack. Anyway, you might want to look into it as the table is quite large and this shouldnt happen under normal operation.

VA2XJM, 2025-03-17:
@aanon4 Yeah, I have not raised the conntrack limit since I updated this node and external (WAN) requests goes above the limits but other than in the logs, there is no impact.

I've already suggested to have this changed in AREDN, but it has not been retained: https://www.pc-freak.net/blog/resolving-nf_conntrack-table-full-dropping-packet-flood-message-in-dmesg-linux-kernel-log/

aanon4, 2025-03-17:
There are 12,500 connections currently going through that node, 11,500 of them appear to be DNS queries of some sort. That seems like a lot. For comparison, my supernode has 2800 active,  2300 are DNS queries; my main home node has 600.

aanon4, 2025-03-17:
Quickly looked back to 2022 and compared it with the code today and it's not changed what it does (could be older than that but couldnt be bothered to look). Not saying what it doest is the right thing, only that the firewall redirections dont look to have altered.

VA2XJM, 2025-03-17:
> Back to the original task ...
> 
> You say this changed for you recently. Can you put an approximate date on that? Or a release version maybe? Would be useful to track this down.

Sorry for the delay. This node was 3.23.12.0 and the GUI was accessible from the mesh. I updated it incrementally up to 3.25.2.0 and then nightly. Once to nightly I disconnected the Ethernet cable and went back to my USB150 and I discovered this bug.

VA2XJM, 2025-03-24:
I took some time tonight to explore the node and I found the below in `/etc/config/firewall`. My understanding is that the upper part with requests coming from `dtdlink` should not be present. I see the same for all port forwarding. 

When the `dtdlink` section is removed and the firewall is restarted (`service firewall restart`), the node GUI is now accessible from the mesh.

So my question is, why is there the dtdlink inserted into the config ?

```
config redirect
        option src dtdlink
        option dest  lan
        option src_dport 80
        option proto tcp
        option src_dip 10.122.185.90
        option dest_ip 10.87.43.77
        option dest_port 80

config redirect
        option src wan
        option dest  lan
        option src_dport 80
        option proto tcp
        option dest_ip 10.87.43.77
        option dest_port 80
```

AI7NC, 2025-05-13:
I have posed the same question in issue [2119](https://github.com/aredn/aredn/issues/2119). For some reason the script behind the gui is creating a redirect entry on 'dtdlink' which isn't needed and in some cases, like with port 80, highly undesirable that breaks node gui connectivity from any mesh device routing through 'dtdlink'. 

We need a fix to remove this behavior.

AI7NC, 2025-05-13:
I'm far from a developer but my suspicion is it's [line 720 in node-setup](https://github.com/aredn/aredn/blob/4f208607e2b12e2bb797af53affbbdac9aa88206/files/usr/local/bin/node-setup#L720).

AI7NC, 2025-05-29:
@aanon4 Can you take a quick look at this, I think it's as simple as removing line 720 above but can't say for sure.

aanon4, 2025-05-29:
That does look correct (with a little bit of whitespace change).

AI7NC, 2025-05-30:
@VA2XJM tomorrow's nightly should have the fix to test

closed 2025-06-11 by aanon4