aredn/aredn #2000: Accessibility improvements?

Orv, 2025-03-18:
A question from a blind ham:

First, thanks to you and the AREDN development team for your tireless work for all of us on the AREDN.  Here's a question for the team.

There are three totally blind operators on the AREDN system in Kansas City.  Could a link be added which could drop the map out of memory for those of us without sight?  The map is not readable by blind mesh operators using accessibility computer software and hardware.  Is the node list available in plain text on the node?  I would suggest if it's possible, to put this in all AREDN releases, so the individual user picks his interface.  The team wouldn't then have to know who was visually impaired, as the user himself should be computer capable of making this choice.  With this choice in ALL AREDN future releases, another group of operators could join the AREDN community.  Is this possible, or, does the node list contingent on the map being on the node?

k1ky, 2025-03-25:
Along this line of thinking, an option to either display in the main GUI or not and have an option to "drill down" to see the map and then exit thus freeing up resources.  Also would like to confirm position changes to avoid accidentally moving the node location.

k1ky, 2025-03-25:
Would "not" displaying the map in the GUI free up any significant memory utilized by the node? If so, how much?

aanon4, 2025-03-25:
@k1ky the map is all done completely in the browser.