aredn/aredn #2011: RouterOS V7.17 and Higher Blocking firmware.

riggs498, 2025-03-20:
Morning,

I am not real sure where this fits. But I have an HAP AX2(which is a HAP AC2). I tried for hours and days to get this loaded with the firmware. I followed the instructions to the letter. You can see it boot and pull the aredn firmware in. It is set to try ethernet first. That was not in the instructions how to make that work. I left it there from 10 minutes to hours at a time. We even went to the movies hoping it would be done on return. Come to find out that the firmware did nothing in the router. Did some reading and it looks like now you have to load firmware from there program and the aredn is not one that is accepted. It also looks like above (I may be off) 15 you can not change the firmware as easy now? Is there away around this? Is there a different way to load this? it looks like the bootloader is blocking the firmware. My router will just reset the routeros that is on it already. 

Ryan

riggs498, 2025-03-20:
> Is not the hAP ax2 an AP using 802.11ax rather than 802.11ac?  The ax family is not in the AREDN supported devices list.
> I put AREDN into a hap ac2 last year with Tiny PXE following the AREDN instructions using try-ethernet-once-then-NAND.  It is in the AREDN doc now.  The RouterOS of that device was 6.49.16 or so when AREDN went in.
> Mikrotik's NetInstall will restore RouterOS to your node.

Anything past 6.49 cannot install a new firmware. Openwrt people are having the same issue. They can only use 6.45 or lower. Whatever is in the bootloader now. Will not let anything install. I cannot find away past it. I am not using the AP on the router. I am just using the router it's self. All wireless is turned off. I want to make a node and a supernode. But that came to a stop very fast.

ab7pa, 2025-03-22:
Did you try the [procedure documented here](https://docs.arednmesh.org/en/latest/arednGettingStarted/installing_firmware.html#mikrotik-first-install-process)?

riggs498, 2025-03-22:
I did, but I will try it again. It really is acting like it is ignoring.

aanon4, 2025-03-25:
We dont support the hAP ax2. It's not an ac2 and has different hardware.

k6ccc, 2025-06-16:
You should be able to use the Mikrotik Netinstall software to restore RouterOS.