aredn/aredn #2020: Cannot do firmware download from server Errror cannot ping server

kp4djt, 2025-03-24:
### Describe the bug
When attempting to update firmware on nodes not connected to internet get "Error cannot connect to server
downloads.arednmesh.org" The node with internet connection WILL upgrade but nodes beyond it will not

- [ ] Does this issue still occur on the latest version? (**Required**)
I was attempting to update from 20250315-a7033633, I have not attempted to manually do upload. 

### Expected behavior
On selecting the new update and clicking on download the fetch bar should scan across then after
a brief bit go to the Node being upgraded screen. 

### Support data
The first support is from the working node with the internet connection

[supportdata-KP4DJT-hAPac3-home-2025-3-24-13-5.tar.gz](https://github.com/user-attachments/files/19435572/supportdata-KP4DJT-hAPac3-home-2025-3-24-13-5.tar.gz)

The second is from one of the up stream nodes that cannot reach the file server

[supportdata-KP4DJT-LHG5ac-2025-3-24-13-6.tar.gz](https://github.com/user-attachments/files/19435582/supportdata-KP4DJT-LHG5ac-2025-3-24-13-6.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Here is what happens when I attempt to ping the file server from both nodes:
Working node:
root@KP4DJT-hAPac3-home:~# ping downloads.arednmesh.org
PING downloads.arednmesh.org (45.63.95.75): 56 data bytes
64 bytes from 45.63.95.75: seq=0 ttl=43 time=55.437 ms
64 bytes from 45.63.95.75: seq=1 ttl=43 time=55.020 ms

Non working node:
root@KP4DJT-LHG5ac:~# ping downloads.arednmesh.org
ping: bad address 'downloads.arednmesh.org'

aanon4, 2025-03-24:
There are so many variables in any given network setup I'll need a lot more information to be able to help. Obviously I checked that I can ping the server (I can) from my nodes (I can) and download an update (I can) so I need to understand what exactly works on your node and what doesnt and how the setups are different. For example, one node can't ping the name but can it ping the IP address? Can it ping any name? If google.com works do other aredn names fail? Please see what you can find out - thanks.

kp4djt, 2025-03-24:
OK going to try and see what I see.

kp4djt, 2025-03-24:
from the good node I first tried the download server addy:
***@***.***:~# ping 45.63.95.75
PING 45.63.95.75 (45.63.95.75): 56 data bytes
64 bytes from 45.63.95.75: seq=0 ttl=43 time=53.686 ms
64 bytes from 45.63.95.75: seq=1 ttl=43 time=53.231 ms
Good.
From the upstream node tried the same thing.
***@***.***:~# ping 45.63.95.75
PING 45.63.95.75 (45.63.95.75): 56 data bytes
^C
--- 45.63.95.75 ping statistics ---
139 packets transmitted, 0 packets received, 100% packet loss

So the IP address is not routing.

The internal URL's are working as I can go to any of those nodes I cannot
update and rout to another node.
Just cannot get to the outside.

kp4djt, 2025-03-24:
Here is a ping to the working node from the node which cannot connect to
the outside:
***@***.***:~# ping KP4DJT-hAPac3-home
PING KP4DJT-hAPac3-home (10.77.51.13): 56 data bytes
64 bytes from 10.77.51.13: seq=0 ttl=64 time=0.960 ms
64 bytes from 10.77.51.13: seq=1 ttl=64 time=0.811 ms

kp4djt, 2025-03-24:
> OK I just loaded 3.25.2.0 into an upstream node. It was  unable to reach
> the outside with Stable loaded.
> This node is part of a node stack that I have built to replace the kludge
> located in a granary here nearby.
> I just plugged the hap ac2 into my lan. It picked up the lan address and
> it is able to reach the server.
>
> Now I am updating it to the later code and will try to see if the nodes
> upstream from it can also upgrade.

The node stack that I built which is under test is linked to my local node
stack over an RF link.
Here is the interesting thing, I just linked out to the Edmond South Node
Stack which is one RF link from
me. I was able to do an upgrade on the ac2 on that system, so why would
that one work and others
fail when they all go to the same access point, the hap ac3 at my home?

OK before sending this I did a bit more checking. The Edmond South Node
Stack has a link to a major
node site in downtown OKC, from there we have a link that goes south to
Norman OK. There is a network
connection there, THAT is how I am getting upgrades for that site. I am
doing traceroute to the internet
router address that is on my home node. As soon as you move away from it
the neighbor nodes do not
know that the route exists.

It would appear that the ac3 is not passing the route info on to the
neighbor nodes. I have the switches
on to make sure that happens.

kp4djt, 2025-03-24:
Did some more looking, if I do a traceroute from the ac3 which is connected
to my home router (192.1689.7.1)
I get the following:Traceroute test started

Source: KP4DJT-hAPac3-home
Target: 192.168.7.1

traceroute to 192.168.7.1 (192.168.7.1), 30 hops max, 46 byte packets
1  192.168.7.1 (192.168.7.1)  0.469 ms

Traceroute done

This is a traceroute from the test mesh, it cannot find an alternate route
so it HAS to try to go to my ac3

Traceroute test started

Source: KP4DJT-HAP-AC2-GRANARY Target: 192.168.7.1 traceroute to
192.168.7.1 (192.168.7.1), 30 hops max, 46 byte packets
1 dtdlink.KP4DJT-LDF5-02 (10.226.161.150) 0.600 ms
2 KP4DJT-LDF5-03 (10.158.54.201) 1.984 ms
3 dtdlink.KP4DJT-hAPac3-home (10.77.51.9) 6.818 ms
4 mid50.KI5VMF-OKLAHOMA (172.31.190.92) 3.648 ms
5 *
6 *

So the node does route to the WAN if the connection is directly from it
internally but anything outside it sends
it to the tunnel server. Even devices plugged into the LAN ports on the
node.

It appears to be that node in particular. Going to do an update and see if
that fixes.

--

Chuck -- KP4DJT
AREDN Meshphone # 405.2005 and 405.2013
HamHotline # 11259
DMR ID 3140958

kp4djt, 2025-03-24:
I did one more test to remove another variable. I used another public
network. I have a cell modem that
is like a sewer pipe, it will take what  your pour down it. I use it to do
mobile AREDN so I KNOW it works.
Anyhow the AC3 does the same thing with that network solution. So it is not
the pubic side that is causing
the issue.

On Mon, Mar 24, 2025 at 4:02 PM Chuck Hast ***@***.***> wrote:

> Did some more looking, if I do a traceroute from the ac3 which is
> connected to my home router (192.1689.7.1)
> I get the following:Traceroute test started
>
> Source: KP4DJT-hAPac3-home
> Target: [...]

-- 

Chuck -- KP4DJT
AREDN Meshphone # 405.2005 and 405.2013
HamHotline # 11259
DMR ID 3140958

kp4djt, 2025-03-24:
As an additional option I plugged the WAN into the hap ac2 that is part of
the stack that will go up on
the granary, this is the node ID if you want to check it out:
KP4DJT-HAP-AC2-GRANARY

When I removed the connection to the WAN port on the ac3 the nodes were
able to upgrade taking the
route to down to Norman OK. So it appears that the issue is with the ac3.
Not sure what to think about
that.

On Mon, Mar 24, 2025 at 5:26 PM Chuck Hast ***@***.***> wrote:

> I did one more test to remove another variable. I used another public
> network. I have a cell modem that
> is like a sewer pipe, it will take what  your pour down it. I use it to do
> mobile AREDN so I [...]

-- 

Chuck -- KP4DJT
AREDN Meshphone # 405.2005 and 405.2013
HamHotline # 11259
DMR ID 3140958

kp4djt, 2025-03-25:
Is there anything else I can do to test this?  It is funny it will not
route the connections to the WAN but it seems
to make the neighbor nodes think it does, if I go out far enough away the
nodes then take the route to a node
30 miles south. That is interesting because it tells me that if a route
fails it will fail over to the next available
path.
I did one more test, the radios that link my site to the Edmond South site,
do as follows. Link radio on my
end when you try to ping or traceroute for example the downloads server, it
fails but as soon as I move to
the radio on the other end of the link it takes the route to the south.  So
the equipment that is on the DtD
net at my end will always try the local node that has the WAN connection
but that will time out. The equipment
which is two hops out will go another path if available

I know this is a lot but I hope this helps resolve this as it is
frustrating.

On Mon, Mar 24, 2025 at 5:41 PM Chuck Hast ***@***.***> wrote:

> As an additional option I plugged the WAN into the hap ac2 that is part of
> the stack that will go up on
> the granary, this is the node ID if you want to check it out:
> KP4DJT-HAP-AC2-GRANARY
>
> When I [...]

-- 

Chuck -- KP4DJT
AREDN Meshphone # 405.2005 and 405.2013
HamHotline # 11259
DMR ID 3140958

kp4djt, 2025-03-25:
One more thing, I just turned off the tunnel now the node also routes out
to the node 30 miles south.
So it appears to be some sort of interaction with the tunnel.

On Mon, Mar 24, 2025 at 7:12 PM Chuck Hast ***@***.***> wrote:

> Is there anything else I can do to test this?  It is funny it will not
> route the connections to the WAN but it seems
> to make the neighbor nodes think it does, if I go out far enough away the
> nodes [...]

-- 

Chuck -- KP4DJT
AREDN Meshphone # 405.2005 and 405.2013
HamHotline # 11259
DMR ID 3140958

kp4djt, 2025-04-04:
Appears to have been a tunnel server issue

closed 2025-04-04 by kp4djt