aredn/aredn #2022: hAP ac2 Adds MeshChat Svc Advertisement with Nightly Build

K5RA, 2025-03-25:
I have a system with two hAP ac lites and one hAP ac2 and a Rocket XW.  One hAP ac lite has Meshchat-api_2.13 and its R-pi service computer has Meshchat_1.02.  The hAP ac2 has Meshchat_2.13.  The two hAP ac lites have VoIP phones.  The hAP ac2 is DtD to the hAP ac lite with Meshchat-api_2.13, and is tunneled to the other hAP ac lite.  The two hAP ac lites are linked by tunnels.  

When I upgrade to the latest NB, the hAP ac2 gets a spurious Meshchat advertisement, like Meshchat-3456.  I have seen this several times, including this morning when I upgraded all three almost simultaneously.  I started with HAP-WH, then HAP2 (the ac2), then HAP-FB (has the service computer).  

Here are screenshots showing the spurious advertisement. 

[image]

[image]

And here is the support data from the hAP ac2. 

[supportdata-K5RA-HAP2-FB-2025-3-25-11-49.tar.gz](https://github.com/user-attachments/files/19452932/supportdata-K5RA-HAP2-FB-2025-3-25-11-49.tar.gz)

Can something be done to eliminate the spurious Meshchat announcements?

K5RA, 2025-03-26:
I reloaded NB 0325 into hAP ac2 that already had NB 0325, and a spurious MeshChat announcement was created.

aanon4, 2025-03-26:
I think this is a problem with MeshChat not AREDN, so please file a bug there. Thanks.

k1ky, 2025-03-27:
May or may not be related - I get random instances of IperfSpeed service showing up on upgraded nodes that have had that service removed on November. It does NOT install IperSpeed, but the service name shows up.

K5RA, 2025-03-27:
As requested by anon4, Bug #34 is opened in kn6plv/meshchat for this item.  
https://github.com/kn6plv/meshchat/issues

closed 2025-03-27 by K5RA