aredn/aredn #2023: 13 of 14 supernode tunnels fail

nc8q, 2025-03-25:
### Describe the bug
I am unable to browse to 13 of 14 of my tunnel partners.

- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
I was expecting to browse to the status page of my partner supernodes

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
Done

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.
I would if I could figure out how to include a screenshot.

### Additional context
Add any other context about the problem here.

nc8q, 2025-03-25:
[supportdata-NC8Q-OHIO-2025-3-25-19-58.tar.gz](https://github.com/user-attachments/files/19455966/supportdata-NC8Q-OHIO-2025-3-25-19-58.tar.gz)

[image]

nc8q, 2025-03-25:
Running nightly 20250325

aanon4, 2025-03-25:
Can you post your attempts to traceroute and ping them please - thanks.

nc8q, 2025-03-27:
traceroute -q 1 -m 3 aa3jc-hapac3-supernode
aa3jc-hapac3-supernode: Name or service not known
Cannot handle "host" cmdline arg `aa3jc-hapac3-supernode' on position 1 (argc 5)
traceroute -q 1 -m 3 af4h-wake-atl-supernode
af4h-wake-atl-supernode: Name or service not known
Cannot handle "host" cmdline arg `af4h-wake-atl-supernode' on position 1 (argc 5)
traceroute -q 1 -m 3 ak4zx-cha-supernode
ak4zx-cha-supernode: Name or service not known
Cannot handle "host" cmdline arg `ak4zx-cha-supernode' on position 1 (argc 5)
traceroute -q 1 -m 3 k2exe-hap-ac3-supernode
k2exe-hap-ac3-supernode: Name or service not known
Cannot handle "host" cmdline arg `k2exe-hap-ac3-supernode' on position 1 (argc 5)
traceroute -q 1 -m 3 k9kdj-indiana-supernode
traceroute to k9kdj-indiana-supernode (10.78.209.132), 3 hops max, 60 byte packets
 1  localnode.local.mesh (10.78.196.97)  0.646 ms
 2  dtdlink.NC8Q-HAP-AC3.local.mesh (10.207.77.99)  1.831 ms
 3  mid1.AC9XU-1 (172.31.93.250)  52.365 ms
traceroute -q 1 -m 3 ke6bxt-lasvegas-supernode
ke6bxt-lasvegas-supernode: Name or service not known
Cannot handle "host" cmdline arg `ke6bxt-lasvegas-supernode' on position 1 (argc 5)
traceroute -q 1 -m 3 ki5vmf-oklahoma-supernode
ki5vmf-oklahoma-supernode: Name or service not known
Cannot handle "host" cmdline arg `ki5vmf-oklahoma-supernode' on position 1 (argc 5)
traceroute -q 1 -m 3 kv3t-chi-supernode
kv3t-chi-supernode: Name or service not known
Cannot handle "host" cmdline arg `kv3t-chi-supernode' on position 1 (argc 5)
traceroute -q 1 -m 3 n2mh-supernode-nj
n2mh-supernode-nj: Name or service not known
Cannot handle "host" cmdline arg `n2mh-supernode-nj' on position 1 (argc 5)
traceroute -q 1 -m 3 n2mh-supernode-oki
n2mh-supernode-oki: Name or service not known
Cannot handle "host" cmdline arg `n2mh-supernode-oki' on position 1 (argc 5)
traceroute -q 1 -m 3 n3fud-pa-supernode
n3fud-pa-supernode: Name or service not known
Cannot handle "host" cmdline arg `n3fud-pa-supernode' on position 1 (argc 5)
traceroute -q 1 -m 3 w8go-oh-co14-supernode01
w8go-oh-co14-supernode01: Name or service not known
Cannot handle "host" cmdline arg `w8go-oh-co14-supernode01' on position 1 (argc 5)
traceroute -q 1 -m 3 wt0f-nfl-supernode
wt0f-nfl-supernode: Name or service not known
Cannot handle "host" cmdline arg `wt0f-nfl-supernode' on position 1 (argc 5)
ping -c 1 aa3jc-hapac3-supernode
ping: aa3jc-hapac3-supernode: Name or service not known
ping -c 1 af4h-wake-atl-supernode
ping: af4h-wake-atl-supernode: Name or service not known
ping -c 1 ak4zx-cha-supernode
ping: ak4zx-cha-supernode: Name or service not known
ping -c 1 k2exe-hap-ac3-supernode
ping: k2exe-hap-ac3-supernode: Name or service not known
ping -c 1 k9kdj-indiana-supernode
PING k9kdj-indiana-supernode.local.mesh (10.78.209.132) 56(84) bytes of data.
64 bytes from K9KDJ-INDIANA-SUPERNODE.local.mesh (10.78.209.132): icmp_seq=1 ttl=62 time=56.2 ms

--- k9kdj-indiana-supernode.local.mesh ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 56.195/56.195/56.195/0.000 ms
ping -c 1 ke6bxt-lasvegas-supernode
ping: ke6bxt-lasvegas-supernode: Name or service not known
ping -c 1 ki5vmf-oklahoma-supernode
ping: ki5vmf-oklahoma-supernode: Name or service not known
ping -c 1 kv3t-chi-supernode
ping: kv3t-chi-supernode: Name or service not known
ping -c 1 n2mh-supernode-nj
ping: n2mh-supernode-nj: Name or service not known
ping -c 1 n2mh-supernode-oki
ping: n2mh-supernode-oki: Name or service not known
ping -c 1 n3fud-pa-supernode
ping: n3fud-pa-supernode: Name or service not known
ping -c 1 w8go-oh-co14-supernode01
ping: w8go-oh-co14-supernode01: Name or service not known
ping -c 1 wt0f-nfl-supernode
ping: wt0f-nfl-supernode: Name or service not known

[image]

[image]

[image]

closed 2025-04-04 by nc8q