aredn/aredn #2037: Supernode tunnels don't decay when gone.

Orv, 2025-03-29:
### Describe the bug
The supernode tunnel from W6BI-SOCAL-SUPERNODE to SK3GA-SM-SUPERNODE has been down for several weeks, yet it still shows in Neighborhood Nodes.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes.
### Expected behavior
A clear and concise description of what you expected to happen.
Would expect nodes at the end of down tunnels to disappear from Neighborhood Nodes fairly promptly.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-W6BI-SOCAL-SUPERNODE-2025-3-29-17-4.tar.gz](https://github.com/user-attachments/files/19522227/supportdata-W6BI-SOCAL-SUPERNODE-2025-3-29-17-4.tar.gz)

aanon4, 2025-03-29:
Wireguard reports that it successfully hand-shook with the far end at some point shortly after boot. I dont know if it's mistaken or not, but it has a valid handshake time about an hour ago (from when the support data was taken). If that was somehow true, or somehow an anomaly with wireguard at boot, I cant tell and have no way to tell. However, if it was just some sort of boot up oddity we wont see another handshake and so, 24 hours from now, the node will be aged out. Let's see if that happens.

Orv, 2025-03-29:
That's about the time I updated the supernode to the latest nightly.   I'll keep an eye on it.  He had reported node problems.  I'll ask him for a support data file from him if available.

Orv, 2025-03-30:
[supportdata-SK3GA-SM-SUPERNODE-2025-3-30-10-59.tar.gz](https://github.com/user-attachments/files/19526676/supportdata-SK3GA-SM-SUPERNODE-2025-3-30-10-59.tar.gz)

Added far end support data file.

aanon4, 2025-03-30:
I dont see it now.

Orv, 2025-03-30:
Yep, it's gone.

closed 2025-03-30 by Orv