aredn/aredn #2040: Tried to revert back to 3.25.2.0 via SSH and bricked node

KenHorse, 2025-03-30:
### Describe the bug
Nightly from 327 caused VERY slow web pages served up on rocket 5AC so tried to revert back to 3.25.2.0 via SSH and bricked node. No RF and no LAN now.

- [ ] Does this issue still occur on the latest version? (**Required**)
No way to tell as cannot communicate with node

### Expected behavior
Expected node to reboot to 3.25.2.0

### Support data
From Nightly 327 before scp'ing 3.25.2.0 binary and running sysupgrade from node's CLI.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context

[supportdata.tar.gz](https://github.com/user-attachments/files/19523557/supportdata.tar.gz)

aanon4, 2025-03-30:
There is no CLI command we document or support to upgrade or downgrade a node. If you use OpenWRT's sysupgrade command it will absolutely brick it.

aanon4, 2025-03-30:
I mean, "works as designed" is not a great designation; but this process is not supported.

KenHorse, 2025-03-30:
What does THAT mean?

aanon4, 2025-03-30:
Nightly builds are test builds. We try to make them work as well as we can. We try to make sure people can revert and upgrade and everything works. However, we *only* support upgrading and downgrading between releases. It would be an unreasonable support task to promise that upgrading and downgrading between arbitrary nightly builds and releases and other nightly builds will always work.  I hope you that makes sense.

KenHorse, 2025-03-30:
What about this?

[https://docs.arednmesh.org/en/latest/arednHow-toGuides/firmware_tips.html](url)

`my-computer:$ scp -P 2222 <aredn-firmware-filename>.bin root@192.168.1.1:/tmp
my-computer:$ ssh -p 2222 root@192.168.1.1
~~~~~~~ after logging into the node with ssh ~~~~~~~
node:# sysupgrade -n /tmp/<aredn-firmware-filename>.bin`

It speaks about upgrading but says nothing (certainly not a warning) about downgrading.

KenHorse, 2025-03-30:
Ok, fair enough

aanon4, 2025-03-30:
Will update the docs. That command might get you back to first boot but it wont keep your state. OpenWRT and AREDN have different places where they keep their state between upgrades/downgrades so it's risky to use their mechanism. There will be a new CLI way to do this in the next release.

aanon4, 2025-03-30:
I dont know what was the problem you had with 307. I had that release on a bunch of nodes, including Rocket AC Lites, without obvious issues.

closed 2025-03-30 by aanon4