aredn/aredn #2041: Nightly 307 on hAP Lite memory hog so reverted to 3.25.2.0 and lost comm with node

KenHorse, 2025-03-30:
### Describe the bug
Nightly 307 on hAP Lite memory hog so reverted to 3.25.2.0 and lost comm with node. Required resorting to first boot which revived it (came up as 3.25.2.0.) but now Wireguard tunnel (client) doesn't work.

- [ ] Does this issue still occur on the latest version? (**Required**)
   As above 3.25.3.0

### Expected behavior
Expect Wireguard tunnel to automatically re-establish itself after time (via NTP) and timezone are properly set

### Support data
From both the client and server nodes. (server node is an AREDN VM running 20250329-a13ccada. N7TB is client, AH6LE is server

### Screenshots
N/A

[supportdata-N7TB-HAP-QTH-2025-3-29-16-12.tar.gz](https://github.com/user-attachments/files/19523568/supportdata-N7TB-HAP-QTH-2025-3-29-16-12.tar.gz)

[supportdata-AH6LE-ProxMoxNode-QTH-2025-3-29-16-12.tar.gz](https://github.com/user-attachments/files/19523569/supportdata-AH6LE-ProxMoxNode-QTH-2025-3-29-16-12.tar.gz)

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-03-30:
Duplicate of this https://github.com/aredn/aredn/issues/1908 where all memory issues are being tracked. We make no promises regarding reverting nightly builds to releases.

closed 2025-03-30 by aanon4

KenHorse, 2025-03-30:
Will moving back to nightly have any benefit at this point?

aanon4, 2025-03-30:
Wait til tomorrow and try again. Yet more space saving then. There's been a fair few since 307.

KenHorse, 2025-03-30:
So when doing a first boot, does that reinitialize memory/config/tables so you're essentially starting with a fresh AREDN install?

aanon4, 2025-03-30:
I'm not sure quite what the question is - sorry. If you either install, reinstall, or return a node to its first boot / first use state, then it's like you just flashed it for the first time out of the box.

KenHorse, 2025-03-30:
That's what I wanted to verify - even though the hAP was downgraded from a nightly to 3.25.2.0, the fact that first boot was run made sure no remnants of that nightly build remained. Still trying to figure out the tunnel failure.....