aredn/aredn #2045: Radio WAN Client Breaks when reassigning 'port1' to different duties

AI7NC, 2025-03-31:
### Describe the bug
If you are using Radio WAN Client mode to obtain internet via wifi, the WAN client irreversibly breaks if you make any changes to port1 via the interface. In my instance I tried both enabling lan as untagged while disabling wan. I also tried changing wan to vlan5 and enabling lan. Both changes irreversibly break wifi connectivity. Changing it back doesn't resolve, only a returning to first boot will allow the WAN wifi connection to work again. It seems the break applies to both radio WAN client via 2.4 or 5.8 GHz. Interface would show 'WAN Client Status' = 'No Connection' despite multiple attempts to disable/enable client WAN wifi. 

I assume the port change is somehow breaking the bridge configuration that's expected for the wifi client to function properly. See below errors from the log that may be relevant. 

Issue observed on 3.25.2.0

### Expected behavior
Reconfiguration of port1 via the web interface should have no effect on radio WAN client connectivity. 

### Support data
These entries below do not show up when the config is working.

	Line 6089: Sun Mar 30 22:43:52 2025 daemon.notice netifd: radio1 (1762): Interface 0 setup failed: BRIDGE_NOT_ALLOWED
	Line 6118: Sun Mar 30 22:43:53 2025 daemon.notice netifd: radio1 (1762): command failed: No such device (-19)

[supportdata-AI7NC-Thurston-AC2-2025-3-30-22-46.tar.gz](https://github.com/user-attachments/files/19528221/supportdata-AI7NC-Thurston-AC2-2025-3-30-22-46.tar.gz)

### Screenshots
Example of one configuration attmempted that breaks radio WAN client connectivity permanently after committing.
 
[image]

### Additional context
Add any other context about the problem here.

aanon4, 2025-03-31:
So when wifi-wan had been broken, did wan work on the port instead? Also, if you changed other port settings but left the wan setting alone, did that break wifi-wan?

AI7NC, 2025-03-31:
I haven't tried either at the remote site, if you need these questions answered before proceeding I can try to replicate via a spare AC2 at home.

aanon4, 2025-03-31:
It would be useful to understand exactly what causes the damage and what doesnt when changing the port settings.

aanon4, 2025-03-31:
https://github.com/aredn/aredn/pull/2046

aanon4, 2025-04-21:
Closing - no confirmation received.

closed 2025-04-21 by aanon4