aredn/aredn #2057: R5AC Works well with 3.25.2 stable but nightlys give low speeds and higher error rates

kp4djt, 2025-04-06:
### Describe the bug
On some of our links running R5AC radios the nightly code just will not give the same results
as the last stable. 

- [ ] Does this issue still occur on the latest version? (**Required**)
- [yes ] 

### Expected behavior
A clear and concise description of what you expected to happen.
We expect similar results to other links on our network. We should be able to
run nightlys on these nodes with similar results to other nodes on our mesh.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
Here are the support data files from the two links in questiion.

[supportdata-AD5JK-PB500AC-DeerCreek-East-2025-4-5-20-4.tar.gz](https://github.com/user-attachments/files/19619128/supportdata-AD5JK-PB500AC-DeerCreek-East-2025-4-5-20-4.tar.gz)

[supportdata-AD5JK-R5AC-EdmondGranary-West-2025-4-5-20-3.tar.gz](https://github.com/user-attachments/files/19619129/supportdata-AD5JK-R5AC-EdmondGranary-West-2025-4-5-20-3.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
These are the two ends running 3.25.2 and after uploading the latest nightly 20250405-981a541e
[DC-EG-3.25.2.odg](https://github.com/user-attachments/files/19619135/DC-EG-3.25.2.odg)

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-04-07:
My browser doesn't know what an ODG file is, so I can read it. Perhaps a zip file?
Also, please include measured performance data (iperf3 for example) rather than just reported rate.
I would also note that nightly and 3.25.2.0 use *exactly* the same kernel and wifi drivers and as these are really the only bits relevant to raw link performance I'm not sure how they can be different.

kp4djt, 2025-04-07:
> My browser doesn't know what an ODG file is, so I cant read it. Perhaps a

Sorry that was supposed to be this pdf. I will have the guys load the
latest nightly back in and run iperf3
test on it.

kp4djt, 2025-04-07:
Here is the iperf3 I did tonight, Aaron has moved the nodes back to Stable.
He will move them back
to nightly and I will get the iperf3 numbers on nightly. What other info
can we supply?

kp4djt, 2025-04-07:
Here is the iperf3 of both ends running latest nightly 20250407-e4281e27

kp4djt, 2025-04-07:
I sure hope we can get this resolved. I just do not know what to think.

aanon4, 2025-04-07:
FYI - you cannot email attachments to a github issue. I think you might be doing that, but it wont work.

kp4djt, 2025-04-08:
OK, do I go into the issue page and add them?

kp4djt, 2025-04-08:
OK I think I see how.

kp4djt, 2025-04-08:
Here are the results of the iperf3 runs
Iperf3 using 3.25.2.0

[DC-EG-3.25.2.pdf](https://github.com/user-attachments/files/19640589/DC-EG-3.25.2.pdf)

Iperf3 run using indicated nightly 

[Iperf3 DC<>GE-Nightly20250407-e4281e27.pdf](https://github.com/user-attachments/files/19640593/Iperf3.DC.GE-Nightly20250407-e4281e27.pdf)

kp4djt, 2025-04-08:
First file was wrong here is the correct one:

[Iperf3 DC<>GE-stable.pdf](https://github.com/user-attachments/files/19640619/Iperf3.DC.GE-stable.pdf)

kp4djt, 2025-04-08:
Just added the files directly, messed up and sent one wrong file but
corrected it

aanon4, 2025-04-08:
Quite why the SNR would be so different (unless that's a time-of-day thing) I'm not sure. Odd.
So .. with release the link runs fast, and with nighty it runs slow. The next test is to have one end of nightly, the other on release, and see if its a specific end on nightly which is causing the issue.

aanon4, 2025-04-08:
Also, please gather support data from both end in all the test configurations. I realize that's quite a bit, but I dont know what I'm looking for yet, so more data is better.

kp4djt, 2025-04-08:
OK going to get it

kp4djt, 2025-04-09:
Here are the files from the test running stable the AD5JK-PB500AC-DeerCreek-East and 20250407-e4281e27 on the AD5JK-R5AC-EdmondGranary-West node

[iperf3-DC-GE.pdf](https://github.com/user-attachments/files/19659661/iperf3-DC-GE.pdf)

[supportdata-AD5JK-PB500AC-DeerCreek-East-2025-4-8-21-50.tar.gz](https://github.com/user-attachments/files/19659678/supportdata-AD5JK-PB500AC-DeerCreek-East-2025-4-8-21-50.tar.gz)

[supportdata-AD5JK-R5AC-EdmondGranary-West-2025-4-8-21-47.tar.gz](https://github.com/user-attachments/files/19659680/supportdata-AD5JK-R5AC-EdmondGranary-West-2025-4-8-21-47.tar.gz)

I had a long day today so was a wee bit late in getting these but I am rested so if you need more, just holler I will be here for a while.

aanon4, 2025-04-09:
Thanks. Could you put both ends back on Release? thanks again.

kp4djt, 2025-04-09:
OK,

kp4djt, 2025-04-09:
OK both ends now on release.

aanon4, 2025-04-09:
Thanks.
So I see with both ends back on release the performance is not what was originally observed and is much closer to that seen with the nightly. Can you remotely power cycle both ends? Specifically *not* just a reboot.

kp4djt, 2025-04-09:
Aaron can power cycle the DeerCreek end the granary end is up on top of a
130 ft granary. We are in the
process of getting all of the command and control down at ground level with
a fibre run up to a switch that
talks to the radios. There will also be power from the solar running up
there on separate power (not PoE)
and then I will be able to do so, until then I will have to climb it.

Aaron noted that the downgrade of the end that was running nightly to
release was very slow to come
back up. I am thinking that I may be able to go over there and find the
breaker that runs power up to the
roof and cycle it but there is a UPS atop and so that frustrates that
attempt. Worse case is one of us goes
over there and climbs it and cycles power.

kp4djt, 2025-04-09:
I wish we had two ends that were easy to deal with but this is the one that
seems to have the issues.

I have Powerbeams at other sites that have good data rates, just these are
the ones in trouble.

kp4djt, 2025-04-09:
Aaron is going to power cycle the Deer Creek end. That will leave the other
end. I can telnet into it if there
is something that can be fiddled that way.

aanon4, 2025-04-10:
Power cycle what's easy (is any of it easy?) and we'll see what that gets us.

kp4djt, 2025-04-10:
We were able to hard reboot the deer creek node. The speed appears to be better. I did both dumps from both ends.

[Iperf3-DC-GE-StableReboot.odt](https://github.com/user-attachments/files/19675875/Iperf3-DC-GE-StableReboot.odt)

[supportdata-AD5JK-PB500AC-DeerCreek-East-2025-4-9-19-23.tar.gz](https://github.com/user-attachments/files/19675893/supportdata-AD5JK-PB500AC-DeerCreek-East-2025-4-9-19-23.tar.gz)

[supportdata-AD5JK-R5AC-EdmondGranary-West-2025-4-9-19-24.tar.gz](https://github.com/user-attachments/files/19675897/supportdata-AD5JK-R5AC-EdmondGranary-West-2025-4-9-19-24.tar.gz)

kp4djt, 2025-04-10:
If  I could get on the DtD port and connect to the managed switch the nodes talk to I could reboot the one that needs a hard reboot, but the switch has a Mikrotik default address of 192.168.88.1 and I cannot reach it being on the DtD port.

aanon4, 2025-04-10:
So this is what I think we know:
With release-to-release the link performance is high.
If the ends are both upgraded, nightly-to-nightly, the link speed is low.
Reverting the links back to release-to-release and the link speed is still low.
Rebooting the nodes in this state doesnt fix anything (I dont believe this was actually tried, so this is speculative, but seems likely)
Power cycling Deer Creek and rebooting Granary restore the speed (pity about the Granary reboot - I really had only wanted a power cycle or nothing).

Next easy thing to try is rebooting both ends and confirming the speed remains high.
Assuming that,
Upgrade Deer Creek to Nightly and see what the speed is (dont touch Granary).
If the speed is low, power cycle Deer Creek (again, doing nothing with Granary) and see what the speed is.

kp4djt, 2025-04-10:
OK, we will get on it.

kp4djt, 2025-04-10:
I am trying to figure out how to route a connection to the managed switch the granary node is connected to. It is DtD so not sure if there is a way to connect to it. If so I would be able to go to the port on the switch that has the node in question connected, and drop PoE then bring it back up.

aanon4, 2025-04-10:
For the experimental plan above, there's no need to power cycle or even reboot Granary.

kp4djt, 2025-04-10:
Going to try to get it tonight.

kp4djt, 2025-04-11:
Dear Creek East is running 20250409-1ccfefd8

[image]

And the granary end 3.25.2.0

[image]

kp4djt, 2025-04-11:
We can now control the granary end. But Aaron got some real gains with the nightly tonight.

kp4djt, 2025-04-11:
No did not but when he upgraded DC to the latest nightly it jumped quite a bit. Prior to the upgrade on DC the rate was about 15mb after  it went up to 21mb as you can see in the two iperf3 screens,

kp4djt, 2025-04-11:
He did neither.  He can do a soft reboot easy.

kp4djt, 2025-04-11:
He just did a soft reboot on them. They came  up nicely. Again the latest nightly seems to be doing much better.

aanon4, 2025-04-11:
Well it looks like you folk have both nodes on nightly and when I run an iperf3 between them I'm getting 20Mbps, which seems at least as good or better than before.

So, our conclusions is ... power cycling the node(s) fixed the performance issues?

kp4djt, 2025-04-11:
That would appear to be the fix.  It is running well now. Of course we will continue to monitor it.

kp4djt, 2025-04-11:
The interesting thing is zero retries, I got on iperf3 on the granary side that had one 3 retry incident the rest were zero in the test. Last night after doing the hard reset on the granary it went to zero and stayed that way and this morning doing several I only got one with one retry entry. What I do see is the speed vary from about 17mb to 22 mb. Is there a way to see how much error correction is being applied to maintain the low retry count?

kp4djt, 2025-04-11:
[image]

Nice and flat.  So now I guess the key is to be able to do a nightly without doing a hard reboot, correct?  I remember  now when you would run these graphs on the old code it would show you a box which showed a speed value. Trying to remember the name of them.

kp4djt, 2025-04-11:
[supportdata-AD5JK-PB500AC-DeerCreek-East-2025-4-11-5-13.tar.gz](https://github.com/user-attachments/files/19703155/supportdata-AD5JK-PB500AC-DeerCreek-East-2025-4-11-5-13.tar.gz)

[supportdata-AD5JK-R5AC-EdmondGranary-West-2025-4-11-5-12.tar.gz](https://github.com/user-attachments/files/19703157/supportdata-AD5JK-R5AC-EdmondGranary-West-2025-4-11-5-12.tar.gz)

aanon4, 2025-04-11:
The actual management of the low level communication (speed, retries, etc.) is a function of the firmware on the radio itself, and is just a proprietary binary blob to us. Nothing we can do about it except report what it does (in the per-neighbor popup). Unsurprising that it will vary over the day due to local conditions.

As for doing an upgrade without this happening again; I dont think that's something we can fix - and of course it has nothing to do with a nightly upgrade because the problem persisted when downgrading to release. It's likely to be a bug in that previously mentioned proprietary binary blob failing to reset some internal hardware we have no access to; nothing we have anyway to change. Not even a unique problem to AREDN or even OpenWRT or this hardware as you can find plenty of reports on people rebooting commercial wifi boxes because they get slow for no reason.

K5RA, 2025-04-11:
So is a node power cycle about once a week a good idea?

aanon4, 2025-04-11:
@k5ra Unless you observer a problem I'm not sure why you would do that.

kp4djt, 2025-04-11:
I recall I think it was on OpenWRT there was (or is have to look) a tool that allowed you to see the constellation pattern of the signal if invoked. Do we have that level of access to be able to do such a thing?

kp4djt, 2025-04-11:
Tim, this morning Aaron KD5JK got home from work and checked the link at 07h22 local. It was horrible, but a few minutes later I looked at it and it was back up to good speed. I wonder if a node with a high gain antenna looking at the horizon where the sun rises or sets might see interference from sun noise? This particular link is east-west and we are right in the time where the sun would be right down the axis of the antenna facing east.  07h22 was about the time the sun came full over the horizon.  We are going to watch it. I am trying to learn how to set up the node exporter so we can graph this and see if the data rate drop follows the sunrise/sunset at this time of year.

aanon4, 2025-04-11:
I've seen constellation tools in other firmware, but dont recall one in OpenWRT. If you have information about one and links to the OpenWRT implementation, feel free to open a feature request for it.

The problem you are KD5JK are now running into might be termed "paying too much attention". Links bounce around all the time but most people dont pay them enough attention to notice. I have an airFiber link which runs between 20 Mbps and 120 Mbps depending on weather and time of day.

kp4djt, 2025-04-11:
See comments below

> I've seen constellation tools in other firmware, but dont recall one in
> OpenWRT. If you have information about one and links to the OpenWRT
> implementation, feel free to open a feature request for it.
>
> The problem you are KD5JK are now running into might be termed "paying too
> much attention". Links bounce around all the time but most people dont pay
> them enough attention to notice. I have an airFiber link which runs between
> 20 Mbps and 120 Mbps depending on weather and time of day.
>
> <https://github.com/aredn/aredn/issues/2057#issuecomment-2798072345>, or
> unsubscribe
>
> I've seen constellation tools in other firmware, but dont recall one in
> OpenWRT. If you have information about one and links to the OpenWRT
> implementation, feel free to open a feature request for it.
I will look for it, but I think I went off in the weeds regarding OpenWRT.
Though it was indicated that
there were tools for such. I will try to follow the rabbit trail.

> The problem you are KD5JK are now running into might be termed "paying too
> much attention". Links bounce around all the time but most people dont pay
> them enough attention to notice. I have an airFiber link which runs between
> 20 Mbps and 120 Mbps depending on weather and time of day.
Yes, I see it all of the time but we saw a lot on this one that was really
up and down, I think KD4JK may have
caught it this a.m. just as the sun was coming up and the east facing radio
got a load of sun noise. It seemed
strange to me that it would run all night like that then at sunrise go off
in the weeds, Of course could be
something else but our favourite star produces a lot of noise down there, I
am going to watch the sunset
and see if we get something similar.

 In a similar vein we are looking at dealing with fading on longpaths.
KE5YZP looking at some of our
planned paths asked me how we were going to deal with  fading, I told her
that was a good question
because back in the days, I saw fading dealt with using space and polar
diversity. She told me she
had the answer and explained. Involves using intelligent switches or
routers you can script your
heart out in and comparing links but certainly sounds workable she told me
she has been doing it for
years. We shall see.

Now the next thing is to learn enough of Grafana to grab the data off of
the nodes and plot the info we
want to monitor.

> <https://github.com/aredn/aredn/issues/2057#issuecomment-2798072345>, or
> unsubscribe

AI7NC, 2025-06-02:
Having grafana grab the node promethueus data is irreplaceable for tracing issues such as this. Here's part of one of our dashboards in the Willamette Valley Mesh Network. 

[image]