aredn/aredn #2065: Some nodes not allowing 6 GHz channels above channel 179

k6ccc, 2025-04-14:
### Describe the bug
I have six different types of nodes that have active AREDN links on 5 GHz.  Several of those node types are not allowing channel assignments above channel 179.  All the nodes are 10 MHz bandwidth.
hAPac3  Channel 179 is max channel on 5 GHz  <Mesh PTMP>
AR750  Channel 179 is max channel  <Mesh Station>
Rocket M5 XM  Channel 184 is max channel (two nodes)  <Mesh>
RB-LHG-5nD  Channel 179 is max channel  <Mesh Station>
Rocket M5 XW  Channel 184 is max channel (two nodes)  <Mesh>
RB-LHG-5HPnD-XL  Channel 184 is max channel on 5 GHz  <Mesh>

I had not realized the pattern until I typed this of the Mesh stations work up to channel 184 and the PTP / PTMP limited to 179.

- [ ] Does this issue still occur on the latest version? (**Required**)
All nodes are NB 20250409-1ccfefd8

### Expected behavior
Work to channel 184 in all cases.

### Support data
Attached for all six node types

[supportdata-K6CCC-RB-LHG-5HPnD-XL-2025-4-14-10-4.tar.gz](https://github.com/user-attachments/files/19740542/supportdata-K6CCC-RB-LHG-5HPnD-XL-2025-4-14-10-4.tar.gz)
[supportdata-K6CCC-Johnstone-SW-5GHz-2025-4-14-10-4.tar.gz](https://github.com/user-attachments/files/19740539/supportdata-K6CCC-Johnstone-SW-5GHz-2025-4-14-10-4.tar.gz)
[supportdata-K6CCC-ROCM5-1-GLENDALE-2025-4-14-10-3.tar.gz](https://github.com/user-attachments/files/19740538/supportdata-K6CCC-ROCM5-1-GLENDALE-2025-4-14-10-3.tar.gz)
[supportdata-K6CCC-RB-LHG-5nD-2025-4-14-10-3.tar.gz](https://github.com/user-attachments/files/19740540/supportdata-K6CCC-RB-LHG-5nD-2025-4-14-10-3.tar.gz)
[supportdata-K6CCC-AR750-Portable-2025-4-14-10-1.tar.gz](https://github.com/user-attachments/files/19740537/supportdata-K6CCC-AR750-Portable-2025-4-14-10-1.tar.gz)
[supportdata-K6CCC-hAPac3-at-Home-2025-4-14-10-0.tar.gz](https://github.com/user-attachments/files/19740541/supportdata-K6CCC-hAPac3-at-Home-2025-4-14-10-0.tar.gz)

aanon4, 2025-04-14:
Are you saying that PtP mode does not allow channels above 179? If so then yes that is a current limitation.

k6ccc, 2025-04-14:
Correct.  OK, so it's a known condition.

aanon4, 2025-04-14:
Yes. I'll convert this to a feature request so I dont forget to look at extending this in the future.

aanon4, 2025-05-20:
https://github.com/aredn/aredn/pull/2157

k6ccc, 2025-05-20:
Great!  I will test this in the next day or two after this version is released.

AI7NC, 2025-05-31:
@aanon4 was this enhancement only applied to the babel-only build? I don't the additional channels for my sector running in 'mesh ptmp' on nightly 20250530.

aanon4, 2025-05-31:
Babel only currently as this is an enhancement rather than a bugfix.

aanon4, 2025-06-11:
Decided I should move this over to the legacy code so people can plan PtXP links.
https://github.com/aredn/aredn/pull/2246