aredn/aredn #2070: 20250418-32ce0382 killed all the tunnels on my X-86 VM

KenHorse, 2025-04-18:
### Describe the bug
Before upgrading, banner warned to backup config, so I did. After upgrade, none of my tunnels were active. They were still defined but not active.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes, 20250418-32ce0382

### Expected behavior
Tunnels should work!

[supportdata-AH6LE-ProxMoxNode-QTH-2025-4-18-13-36.tar.gz](https://github.com/user-attachments/files/19816162/supportdata-AH6LE-ProxMoxNode-QTH-2025-4-18-13-36.tar.gz)

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

KenHorse, 2025-04-18:
I restored config backup but no change

aanon4, 2025-04-18:
Upgraded from what version?

aanon4, 2025-04-18:
The WAN interface has a 10.X address? That's seems problematic.

KenHorse, 2025-04-18:
Not what the home page shows

[image]

closed 2025-04-18 by aanon4

K5RA, 2025-04-19:
One of our mesh island nodes upgraded to NB 0418 this morning.  That knocked out the tunnel that I serve to him.  Rebooting his node (tunnel client end) did not fix it.  Rebooting the my node (tunnel-serving end) fixed it.  Mesh island has been solid so far today.