aredn/aredn #2071: New ac3 has no WAN DNS entries, no Ethernet port lights, can't ping default gateway

Orv, 2025-04-19:
### Describe the bug
A brand-new ac3 - loaded properly, but has no entries for WAN DNS, Ethernet Port LEDs don't light, and can't ping default gateway.

- [ ] Does this issue still occur on the latest version? (**Required**)
Showed issue with original load of 3.256.2.0 and also with load of 20250418

### Expected behavior
Normal behavior for an ac3  :-)

### Support data

[supportdata-N7UV-QTH-2025-4-19-23-33.tar.gz](https://github.com/user-attachments/files/19823212/supportdata-N7UV-QTH-2025-4-19-23-33.tar.gz)

### Screenshots

[image]

[image]

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-04-20:
I had someone with a similar problem with a hAP ac2 just yesterday. Same symptoms and same errors. It was a device I gave him too. He flashed it several times and eventually it worked fine as expected. We dont know why it happened or why it eventually worked.

So let's try this. Reflash from scratch - RAM boot then sysupgrade upload. Please keep both those exact files because, if it fails again, I want those exact files (not a copy which may or may not be identical).

With the node sat waiting for the node name and password to be set, telnet/ssh into it and generate and download the support data. Next type `logread;logread -f` and let it sit running. Back at the browser go through the process of setting a name and password. There'll be some output from the logging commands above, so please capture all of that.

See how the node appears once it's rebooted and, assuming it's messed up, generate more support data.

Package all that up, the images, support datas, log capture, as a care package for me :-)

Orv, 2025-04-20:
[image]

[image]

[supportdata.tar.gz](https://github.com/user-attachments/files/19823370/supportdata.tar.gz)

After doing a fresh reload from scratch, ac3 started working a bit better:  WAN DNS entries are there.
   Support data file is from prior to configuring node.

However, all IP address are showing 10.x.x.x - the default gateway should be 192.168.1.1 (user doesn't use the 10 network at home)

[image]

Here's a support data file post-configuration.

[supportdata-N7UV-QTH-2025-4-13-17-30.tar.gz](https://github.com/user-attachments/files/19823391/supportdata-N7UV-QTH-2025-4-13-17-30.tar.gz)

Orv, 2025-04-20:
Reloading from scratch, rebooting house DHCP server, then rebooting ac3 resolved all known issues.

closed 2025-04-20 by Orv