aredn/aredn #2074: Supported devices Updates

Orv, 2025-04-22:
1.  Somehow indicate in the entry for hAP ac lite and hAP ac lite TC that while supported, they are no longer recommended.
2. Add the Vultr platform to the x86 / Virtual Machine table.

K5RA, 2025-04-23:
Orv - can you elaborate on what motivates this enhancement for the two hAPs?  
Thanks.

Orv, 2025-04-23:
Here's the deal, Tim - the hAP ac Lite is the oldest, busiest device in the AREDN supported devices arena.  What with the code growing with the addition of Babel, and maybe serving a bunch of tunnels, the ac lite is a bit memory challenged.   And it turns out that the WAP client in the OpenWrt code is a memory hog - kind of the last straw for the ac Lite.  It runs out of memory and stops behaving as expected.
There are some workarounds (don't run a hotspot and minimize tunnels), but for now the easiest solution is update to an ac3

aanon4, 2025-04-24:
https://github.com/aredn/aredn/pull/2080

closed 2025-05-09 by aanon4