aredn/aredn #2077: > 144.4 Mbps on Amateur Radio channels

nc8q, 2025-04-23:
**Is your feature request related to a problem? Please describe.**
Mikrotik: SXT-5-AC  and LHGG-5-AC-XL.
I would like to explore 40 MHz band width channels in hopes of > 144.4 Mbps links.
Currently 40 MHz band width on channel 36 can do 300 Mbps.
Currently 40 MHz band width on channel 140 can do only > 144.4 Mbps.

**Describe the solution you'd like**
I would like > 144.4 Mbps on Amateur Radio channels

Attaching DataSupport and /status images

[supportdata-NC8Q-SXT-5-AC-RED-2025-4-23-10-15.tar.gz](https://github.com/user-attachments/files/19869189/supportdata-NC8Q-SXT-5-AC-RED-2025-4-23-10-15.tar.gz)

[supportdata-NC8Q-LHG-5AC-XL-2025-4-23-10-14.tar.gz](https://github.com/user-attachments/files/19869192/supportdata-NC8Q-LHG-5AC-XL-2025-4-23-10-14.tar.gz)

[image]

[image]

[image]

[image]

aanon4, 2025-04-24:
The support data looks like it's when the channel is 36, so it all looks great. Could you add support data for channel 140? Thanks.

nc8q, 2025-04-24:
Channel 124

[supportdata-NC8Q-LHG-5AC-XL-2025-4-24-8-23_ch124.tar.gz](https://github.com/user-attachments/files/19887323/supportdata-NC8Q-LHG-5AC-XL-2025-4-24-8-23_ch124.tar.gz)
[supportdata-NC8Q-SXT-5-AC-RED-2025-4-24-8-21_ch124.tar.gz](https://github.com/user-attachments/files/19887324/supportdata-NC8Q-SXT-5-AC-RED-2025-4-24-8-21_ch124.tar.gz)

nc8q, 2025-04-24:
Channel 140

[supportdata-NC8Q-SXT-5-AC-RED-2025-4-24-11-4_ch140.tar.gz](https://github.com/user-attachments/files/19889956/supportdata-NC8Q-SXT-5-AC-RED-2025-4-24-11-4_ch140.tar.gz)
[supportdata-NC8Q-LHG-5AC-XL-2025-4-24-10-52_ch140.tar.gz](https://github.com/user-attachments/files/19889957/supportdata-NC8Q-LHG-5AC-XL-2025-4-24-10-52_ch140.tar.gz)