aredn/aredn #2081: hAP ac2 Mesh Radio Max Distance Always Zero

K5RA, 2025-04-24:
### Describe the bug
A clear and concise description of what the bug is.

- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
A clear and concise description of what you expected to happen.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-04-24:
You mention two nodes but only added support data for one of them. Please provide both as you're comparing the two configurations. Thanks.

K5RA, 2025-04-24:
I did an experiment this morning: powered off the SXT.  That made no difference.  I will power off both the SXT and the Rocket M2 XW and see what that does.

aanon4, 2025-04-24:
Without support data from these experiments I can't do much I'm afraid.

K5RA, 2025-04-24:
I understand.  Just a few distractions today. 

New experiment.  The hAP ac2, HAP2-FB, is DtD to HAP-FB, a hAP ac lite with four tunnel servers and four tunnel clients.  HAP2-FB is tunneled to HAP-WH.  HAP2-FB, the hAP ac2, now has no attached nodes.  

Full Status screen screenshot of HAP2-FB.  

[image]

Support data for HAP2-FB.

[supportdata-K5RA-HAP2-FB-2025-4-24-12-39.tar.gz](https://github.com/user-attachments/files/19897128/supportdata-K5RA-HAP2-FB-2025-4-24-12-39.tar.gz)

Screenshot of HAP2-FB's Radios & Antennas dialogue box.  Max distance setting is 4.2 miles.  Data on Status screen shows zero miles.

[image] 

Support data for HAP-FW and HAP-WH.  

[supportdata-K5RA-HAP-FB-2025-4-24-12-39.tar.gz](https://github.com/user-attachments/files/19897197/supportdata-K5RA-HAP-FB-2025-4-24-12-39.tar.gz)

[supportdata-K5RA-HAP-WH-2025-4-24-12-39.tar.gz](https://github.com/user-attachments/files/19897200/supportdata-K5RA-HAP-WH-2025-4-24-12-39.tar.gz)

Question regarding your comment about adjacent nodes:  Does the Node Status screen max distance value reflect what is in the dialogue box, or reflect the distance to the nearest local or neighbor node ?   I thought that might be the case, but powering down the SXTsq 5 ac node changed nothing on the max distance value of the 5 GHz Mesh radio in the hap ac2.

aanon4, 2025-04-24:
We've updated the docs in a couple of places to try to make this clearer.

K5RA, 2025-04-24:
I powered up the Rocket M2 XW.  It is DtD to HAP2-FB.  The Rocket is running NB 20250424.  A WiFi Scan on the Rocket shows only one signal.  Since SNR is 95, I assume it is the Rocket itself. 

The Max Distance value on the Rocket's Node Status screen and its Radios & Antennas dialogue box agree.  Why is the Rocket different from the HAP2 node's 5 GHz Mesh radio? 

RK2C Radios & Antennas dialogue box showing 9.5 miles with RK2C Status screen in background showing 10 miles.

[image]

RK2C support data.

[supportdata-K5RA-RK2C-FB-2025-4-24-14-20.tar.gz](https://github.com/user-attachments/files/19898021/supportdata-K5RA-RK2C-FB-2025-4-24-14-20.tar.gz)

K5RA, 2025-04-24:
I see the AREDN Docs edit (#490) describing the Status Screen display.  That should help.

The write-up in AREDN Docs/Getting Started Guide/Node Admin Guide/Mesh Settings/Radios and Antennas/Maximum Distance is clear IF you understand that the max distance value in the dialogue box and that on the Status Screen have different definitions. 

I feel that is happening with my RK2C-FB node is not clear based on these definitions.  

New RK2C-FB Radios & Antennas screenshot with new value of max distance showing in both Status and dialogue box screens. 

[image]

RK2C-FB Support Data

[supportdata-K5RA-RK2C-FB-2025-4-24-14-51.tar.gz](https://github.com/user-attachments/files/19898290/supportdata-K5RA-RK2C-FB-2025-4-24-14-51.tar.gz)

K5RA, 2025-04-24:
Another experiment.  Because the hAP ac nodes do not provide 5 MHz bandwidth, I changed RK2C-FB's bandwidth to 10 MHz.  Then I turned off the 5 GHz Mesh radio in HAP2-FB and turned on a 2.4 GHz Mesh radio.  After the commit and a brief coming-up, the Max Distance in HAP2-FB's Status screen showed a max distance of zero, as expected because there is a linking node whose distance is very small (about 1 meter).  The Max Distance in the dialogue box is 8.2  miles.  

RK2C-FB screenshot.  

[image]

Now...how to explain these cases in AREDN Docs?

aanon4, 2025-04-26:
I was talking to someone about this and they reminded me that some radios, notably those which sit on your desk and not up a tower, wont let you set the distance and will report their preset value, usually 0, regardless. The hAP ac2 is one such radio. What we display on the main page is what the radio is doing regardless of what you try to tell it to do.

Given this, which radios do you see this problem on?

K5RA, 2025-04-26:
I have the   hAP ac2   configured with a 5 GHz Mesh radio, and an   SXTsq 5 ac   linked to it.  The SNR is 60dB because they are only a few feet apart.  I gave the SXT a bogus latitude location 0.05 deg south of where it really is to produce a 3.5 mile distance in the Neighbors displays of both the hAP ac2 and the SXT.  Both nodes show max distance of zero miles on the Status display.  

Also on my bench is a   Rocket M2 XW   which is DtD to the  hAP ac2 .   It has Max Distance set of 22.2 miles and displayed in Status  of 23 miles.  The Rocket has no one to talk to.

aanon4, 2025-04-26:
So the 0 is only on Mikrotik AC radios?

K5RA, 2025-04-26:
The SXTsq 5 ac does it as well.

aanon4, 2025-04-26:
https://github.com/aredn/aredn/issues/2082

K5RA, 2025-04-26:
Enhancement.  Will be addressed in future release.

closed 2025-04-26 by K5RA

aanon4, 2025-04-26:
Documented current behavior.