aredn/aredn #2087: Nodes Flapping

jltaft1962, 2025-05-06:
### Describe the bug
A clear and concise description of what the bug is.
Our AREDN system supports DMR repeater sites as backup to regular internet. Lately 3 sites have been flapping. All nodes are running 3.25.2.0 firmware.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes, we tried the latest nightly build which is when the flapping started. We then reverted to 3.25.2.0 trying to fix it.

### Expected behavior
A clear and concise description of what you expected to happen.
We expect the network sites to stay connected to one another.  DMR traffic is less than 200kbits/s, so loading should not be a problem.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**) This node covers 2 of the 3 sites and 3 of 4 repeaters. It was recorded when the nodes were offline as indicated by them being missing from the c-bridge dashboard.

[supportdata-N1MVA-Uncanoonuc4-NH-TUN-2025-5-6-17-14.tar.gz](https://github.com/user-attachments/files/20070778/supportdata-N1MVA-Uncanoonuc4-NH-TUN-2025-5-6-17-14.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-05-06:
If I can summaries. The flapping was first observed after you upgraded one (or more?) nodes to a nightly and reverting it did not stop the flapping - is that correct?
The provided support data is for a node without any configured mesh radio, so I dont see anything about potential link problems. Also, the logs is full of the following messages:
```
Tue May  6 00:17:13 2025 daemon.info olsrd[12881]: Skipped 619 mid duplicate warnings.
Tue May  6 00:17:33 2025 daemon.info olsrd[12881]: Node 10.25.241.173 is publishing your ip 10.25.241.169 as MID!
Tue May  6 00:17:40 2025 daemon.info olsrd[12881]: Node 10.25.241.173 is publishing your ip 10.25.241.169 as MID!
Tue May  6 00:17:40 2025 daemon.info olsrd[12881]: Node 10.25.241.173 is publishing your ip 10.25.241.169 as MID!
Tue May  6 00:17:40 2025 daemon.info olsrd[12881]: Node 10.25.241.173 is publishing your ip 10.25.241.169 as MID!
Tue May  6 00:19:18 2025 daemon.info olsrd[12881]: Node 10.25.241.173 is publishing your ip 10.25.241.169 as MID!
Tue May  6 00:19:19 2025 daemon.info olsrd[12881]: Node 10.25.241.173 is publishing your ip 10.25.241.169 as MID!
Tue May  6 00:19:19 2025 daemon.info olsrd[12881]: Node 10.25.241.173 is publishing your ip 10.25.241.169 as MID!
Tue May  6 00:19:20 2025 daemon.info olsrd[12881]: Node 10.25.241.173 is publishing your ip 10.25.241.169 as MID!
Tue May  6 00:19:27 2025 daemon.info olsrd[12881]: Node 10.25.241.173 is publishing your ip 10.25.241.169 as MID!
```
Which isn't good :-(
10.25.241.169 is the address of your dtdlink interface. 10.25.241.174 is your main IP address, but it's odd that the node somewhere on your network causing routing problems has an IP address which is only one different, 10.25.241.173.

Anyway, you need to track down this node and work out what's going on there.

jltaft1962, 2025-05-06:
Thanks, Tim.The upgraded nodes were two Mikrotik ac3 routers, one at a dmr site and one at my home. They are connected by wireguard tunnel for network management. When the dmr repeaters began flapping, I reverted them to the previous production version. We’ll check out x.x.x.173.- Jay