aredn/aredn #2088: Pair of OpenWRT-One devices do not mesh

wu2s, 2025-05-07:
### Describe the bug
A clear and concise description of what the bug is.
A pair of the new OpenWRT One devices fails to mesh with each other on either of the 2.4 GHz or 5.6 GHz bands.

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] The AREDN version on both of the units is NB 20250429
- [ ] 

### Expected behavior
A clear and concise description of what you expected to happen.
Expected that the devices mesh with each other automatically when the channel, bandwidth and SSID are the same..

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
Three (3) examples of each pairing are attached.

[supportdata-WU2S-OWRT1-A-2025-5-6-18-57.tar.gz](https://github.com/user-attachments/files/20074042/supportdata-WU2S-OWRT1-A-2025-5-6-18-57.tar.gz)
[supportdata-WU2S-OWRT1-A-2025-5-6-19-5.tar.gz](https://github.com/user-attachments/files/20074046/supportdata-WU2S-OWRT1-A-2025-5-6-19-5.tar.gz)
[supportdata-WU2S-OWRT1-A-2025-5-6-19-12.tar.gz](https://github.com/user-attachments/files/20074044/supportdata-WU2S-OWRT1-A-2025-5-6-19-12.tar.gz)
[supportdata-WU2S-OWRT1-B-2025-4-18-11-48.tar.gz](https://github.com/user-attachments/files/20074043/supportdata-WU2S-OWRT1-B-2025-4-18-11-48.tar.gz)
[supportdata-WU2S-OWRT1-B-2025-4-18-11-51.tar.gz](https://github.com/user-attachments/files/20074047/supportdata-WU2S-OWRT1-B-2025-4-18-11-51.tar.gz)
[supportdata-WU2S-OWRT1-B-2025-4-18-12-2.tar.gz](https://github.com/user-attachments/files/20074045/supportdata-WU2S-OWRT1-B-2025-4-18-12-2.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-05-07:
Can they connect to anything else over wifi? Let's stick with 5GHz for simplicity. I see it scan lots of other nodes in your vacinity but they're all on different frequencies. If you pick a non-One device you have setup, will it connect to those?

aanon4, 2025-05-07:
Ah - all the other stuff I see looks like it just might be non-AREDN stuff in your house. I'd still be interested if it can connect to anther AREDN radio.

ab7pa, 2025-05-07:
They would have to use 20 MHz width, right?

aanon4, 2025-05-07:
That's correct.
I spun up my One and could not get it to mesh with another radio, a rocket ac in this case. Trying PtMP mode now.

aanon4, 2025-05-07:
PtMP (and PtP by implication) works fine. So just looks like Mesh (adhoc) is not functional.

wu2s, 2025-05-07:
Yes they have to use 20 MHz bandwidth because that is the narrowest value available on the OpenWRT ONE.
I left the OpenWRT-B unit on channel 137 at 20 MHz bandwidth. This unit operates by itself with no DtD devices attached.
I changed WU2S-H3-X to the same channel and bandwidth. This unit is in another room.
After waiting for over 5 minutes, the 2 units do not mesh.

aanon4, 2025-05-07:
We'll mark this as a known issue for the upcoming release and then block selecting basic mesh as a radio option after that.