aredn/aredn #2090: NAT LAN netmasks bigger than /24 and addresses not at start of subnet not saved

rjacoby00, 2025-05-07:
### Describe the bug
When entering a LAN subnet in NAT mode larger than /24, the input is changed to /24 once exiting the window and no error is thrown. When entering a LAN address in NAT mode, the input is changed to the first address in the CIDR block and no error is thrown.

- [x] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
LAN subnets bigger than /24 (ex. 192.168.136.0/23) should be allowed.
LAN addresses should not be constrained to the first address in CIDR block

### Support data
[supportdata-W6BHZ-Shack-To-Cuesta-P2P-2025-5-7-4-48.tar.gz](https://github.com/user-attachments/files/20076375/supportdata-W6BHZ-Shack-To-Cuesta-P2P-2025-5-7-4-48.tar.gz)

### Screenshots
Before exiting Network dialogue 

[image]

After exiting Network dialogue

[image]

aanon4, 2025-05-07:
A question and a thing to try:
Question - does the IP Address box not get highlighted in red for this configuration? It does in my browser.
To try - can you try 255.255.252.0 as a netmask? Might just be a fault with 255.255.254.0 but would like to verify what works for you.

rjacoby00, 2025-05-08:
Same behavior with 255.255.252.0.  It doesn't highlight the boxes in red for either case while LAN DHCP is off, but does while DHCP is on. I didn't test it before and I didn't think it was related, but while DHCP is enabled it seems to remember the configuration fine.

aanon4, 2025-05-09:
Fixed