aredn/aredn #2101: Mystery Link Loss After Upgrade to 3.25.5.0

K5RA, 2025-05-08:
My K5RA-HAP-FB node serves a tunnel to N5OM-HAP1.  I upgraded my HAP-FB to new release 3.25.5.0 about an hour after it became available.  All four tunnel servers and four tunnel clients appeared to come back OK.  Later in the day N5OM upgraded his two nodes to the stable release, and the tunnel from me to him was not restored.  Here is the Node Status screen shot and support data on my end.  

[image]

[supportdata-K5RA-HAP-FB-2025-5-7-15-42.tar.gz](https://github.com/user-attachments/files/20104233/supportdata-K5RA-HAP-FB-2025-5-7-15-42.tar.gz)

I was able to reach N5OM-HAP1, probably via a back-up route through KI5ZSF-HAP1 and N5OM-HAP2, and get a screen shot.  His tunnel client to me was inactive, and my node did not show as a neighbor on his HAP1.  

[image]

I rebooted K5RA-HAP-FB, and the inactive tunnel was restored.  The free RAM on my HAP1 node increased from about 7.5MB to about 9.6MB after the reboot. 

This mystery loss of one or more tunnels has happened several times in the last few weeks.  Usually a node on one end of a tunnel shows significant error rate and node on the other end shows 0%.  Rebooting the   node showing errors   fixes link.  Is this an effect of Babel/OLSR interactions, am I not waiting long enough for tunnels to settle, or am I asking too much of my hAP ac lite?

aanon4, 2025-05-08:
I dont suppose you have a way to reliably reproduce this?
There are a smattering of reports of similar things that I'm seeing. It looks to be some sort of race condition where one end becomes convinced the other isnt there despite receiving traffic from it. Restarting Babel (rebooting being the easiest way) on either end fixes the problem.
If you have a reliably way to make this happen then please let me know as it'll help resolve this.

aanon4, 2025-05-08:
So I've rolled out a fix for this which will be in tomorrow's nightly build. I would appreciate greatly if you could test it. This one is very hard to reproduce but I can occasionally make it happen by continually rebooting my local supernode ... which is hardly the best testing environment for lots of reasons, but it has been the only semi-reliable way I can make it happen.  The change fixes the problem there, but more people trying it would be appreciated.

aanon4, 2025-05-08:
https://github.com/aredn/aredn_packages/pull/104

aanon4, 2025-05-09:
Did you have chance to verify today's nightly build?

K5RA, 2025-05-10:
We had a recurrence of a tunnel failure.  I have screenshots, and the support data from my end. 

I had NB 0509 in my hAP ac Lite.  N5OM upgraded from 3.25.5.0 to NB 0509, and the tunnel between us went inactive.  I think the tunnel between his two ac Lites also went inactive.  Rebooting my end, the tunnel server to him, fixed that tunnel.  I will get that data to you after I have coffee this morning.

I also have a tunnel failure between my two SXTsq 5 acs.  Had to reboot both to get near-zero error rates.  They both have NB 0509.  I have support data for that event as well. 

I appreciate all you are doing to make this significant technological advancement (AREDN Babel) possible.

K5RA, 2025-05-10:
My test bed has my HAP-FB and HAP-TC-FB (hAP ac Lites), and HAP2-FB (a hAP ac2) all with various tunnels and 5GHz mesh radios.  Also two SXTsq 5 ac nodes linked to the hAPs.  Also a Rocket M2 XW. 

My hAP nodes are running NB 20250509.  When N5OM upgraded his N5OM-HAP1 node from 3.25.5.0 to NB 0509, the tunnel served by HAP-FB to N5OM-HAP1 went inactive.  We saw this same event 07 May when he upgraded to 3.25.5.0.  Rebooting my HAP-FB re-activated the tunnel. 

Here are support data from my node and both ends Node Status screenshots from the event yesterday before rebooting my HAP-FB node (the tunnel server).  K5RA-HAP-FB does not show as a neighbor to N5OM-HAP1.

[supportdata-K5RA-HAP-FB-2025-5-9-21-55.tar.gz](https://github.com/user-attachments/files/20139709/supportdata-K5RA-HAP-FB-2025-5-9-21-55.tar.gz)

[image]

[image]

Here are Node Status screenshots after rebooting HAP-FB. 

[image]

[image]

aanon4, 2025-05-10:
Thanks for this.
For screenshots what I would really like is the detailed popup for the specific failed link. It has a lot more detail there. If the tunnel is still failed and you can attach them, that'd be super helpful.

aanon4, 2025-05-10:
Well on the plus side it looks to be a different problem from the one I've been debugging. Here's the top few lines of output from the app `babel-dump`:
```
BABEL 1.0
version unknown
host K5RA-HAP-FB
my-id 78:9a:18:ff:fe:4d:16:fc
ok
add interface br-dtdlink up true ipv6 fe80::97:d1ff:fe64:c4d5 ipv4 10.77.22.248
add interface wgs1 up true ipv6 fe80::200:acff:fe1f:11b7 ipv4 172.31.17.183
add interface wgs3 up true ipv6 fe80::200:acff:fe1f:ee3 ipv4 172.31.14.227
add interface wgc3 up true ipv6 fe80::200:acff:fe1f:be34 ipv4 172.31.190.52
add interface wgs2 up true ipv6 fe80::200:acff:fe1f:cbd7 ipv4 172.31.203.215
add interface wgs0 up true ipv6 fe80::200:acff:fe1f:2dc9 ipv4 172.31.45.201
add interface wgc0 up true ipv6 fe80::200:acff:fe1f:be36 ipv4 172.31.190.54
add interface wgc1 up true ipv6 fe80::200:acff:fe1f:be38 ipv4 172.31.190.56
add interface wgc2 up true ipv6 fe80::200:acff:fe1f:be3a ipv4 172.31.190.58
add interface wlan0 up true ipv6 fe80::7a9a:18ff:fe4d:16fc ipv4 10.77.22.252
add neighbour 77ece010 address fe80::200:acff:fe1f:ee2 if wgs3 reach ffff ureach 0000 rxcost 306 txcost 306 rtt 28.086 rttcost 24 cost 330
add neighbour 77ecc410 address fe80::d601:c3ff:fe51:e54 if wlan0 reach ffff ureach 0000 rxcost 256 txcost 264 cost 264
add neighbour 77ecc550 address fe80::d601:c3ff:fea8:aa5a if wlan0 reach ffff ureach 0000 rxcost 256 txcost 273 cost 273
add neighbour 77ecc2a0 address fe80::f61e:57ff:fea0:f25e if wlan0 reach ffff ureach 0000 rxcost 256 txcost 273 cost 273
add neighbour 77ecc690 address fe80::f61e:57ff:fea0:f1a1 if wlan0 reach ffff ureach 0000 rxcost 256 txcost 256 cost 256
add neighbour 77ecc160 address fe80::200:acff:fe1f:be3b if wgc2 reach ffff ureach 0000 rxcost 306 txcost 306 rtt 19.612 rttcost 13 cost 319
add neighbour 77ecc020 address fe80::200:acff:fe1f:be39 if wgc1 reach ffff ureach 0000 rxcost 306 txcost 306 rtt 26.784 rttcost 22 cost 328
add neighbour 77ece680 address fe80::200:acff:fe1f:be37 if wgc0 reach ffff ureach 0000 rxcost 306 txcost 306 rtt 20.494 rttcost 14 cost 320
add neighbour 77ece400 address fe80::200:acff:fe1f:2dc8 if wgs0 reach ffff ureach 0000 rxcost 306 txcost 306 rtt 23.045 rttcost 17 cost 323
add neighbour 77e07d80 address fe80::200:acff:fe1f:11b6 if wgs1 reach ffff ureach 0000 rxcost 306 txcost 306 rtt 30.801 rttcost 28 cost 334
add xroute 10.104.183.224/29-0.0.0.0/0 prefix 10.104.183.224/29 from 0.0.0.0/0 metric 0
add xroute 10.77.22.248/32-0.0.0.0/0 prefix 10.77.22.248/32 from 0.0.0.0/0 metric 0
add xroute 10.77.22.252/32-0.0.0.0/0 prefix 10.77.22.252/32 from 0.0.0.0/0 metric 0
add xroute 10.104.183.225/32-0.0.0.0/0 prefix 10.104.183.225/32 from 0.0.0.0/0 metric 0
```
The failed tunnel is `wgc3`. In the above Babel knows about the interface and thinks it is up and active. But, in the `add neighbor` section there are no neighbors associated with this interface.

The tunnels does appear to be up, but looking at LQM it cannot ping the node on the other end. If I check the Wireguard handshake times, it hasnt had a successful handshake with the other end in some time. Also, good old OLSR hasnt manage to contact the other end either.

So .. summary so far .. I'm not sure what's going on with this tunnel, but it looks like wireguard is failing to pass traffic rather than it being a Babel issue.

K5RA, 2025-05-10:
All my nodes are reporting 0% error rates on all links.  I have been fighting with my nodes with mesh radios since late last night trying to fix high error rates.  I think they were all OK at some point after NB 0509 install yesterday morning.  After I brought up the SXT2 mesh radio and disconnected the HAP-FB/HAP2 DtD link (to challenge my test bed (!) ), I started to see non-zero error rates on other RF links.  

I noticed this morning that N5OM-HAP1 node displays GMT (Z), and I display US Central (R).  Also N5OM-HAP1 has two tunnels (other than his HAP2).  When he upgraded, the one I served went down, the other stayed up.   This is the same as the 07 May event.  So maybe it is specific to the WireGuard link between us. 

How do I bring up the detailed pop-up for a specific tunnel?

K5RA, 2025-05-10:
Is this the tunnel pop-up you would like to have.   This is of an "excellent" link. 

[image]

aanon4, 2025-05-10:
That's the one. And it is indeed excellent right now. The earlier support data showed a ping success of 0% and babel routes and olsr routes were both 0. No longer it seems. Did anything obvious change?

K5RA, 2025-05-10:
Not that I know of.  The link reported 50% error rate at 0256Z, and 0% at 0303Z after I rebooted my HAP-FB, the tunnel server.

K5RA, 2025-05-15:
It seems that 3.25.5.1 has done the job on this one.  In our mesh island, a majority of the nodes are now running 3.25.5.1 or a later NB.  I had a strange link loss for a few hours (about 0300-0800? this morning) on an RF link between two physically adjacent nodes, but otherwise, link reliability has been excellent.

closed 2025-05-15 by K5RA