aredn/aredn #2102: Pair of OpenWRT One device do not connect in infrastructure mode

wu2s, 2025-05-08:
### Describe the bug
A clear and concise description of what the bug is.
One dive is a Station and the other is a PtP unit.
They do not connect.

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes, both are running NB 20250508

### Expected behavior
A clear and concise description of what you expected to happen.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-WU2S-OWRT1-A-2025-5-8-13-10.tar.gz](https://github.com/user-attachments/files/20108718/supportdata-WU2S-OWRT1-A-2025-5-8-13-10.tar.gz)
[supportdata-WU2S-OWRT1-B-2025-5-8-13-9.tar.gz](https://github.com/user-attachments/files/20108719/supportdata-WU2S-OWRT1-B-2025-5-8-13-9.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-05-08:
Could you try putting the PtP one in PtMP mode and compare? Thanks.

wu2s, 2025-05-08:
Changed unit A to PtMP and wait for at least 15 minutes.
Still no connections.

[supportdata-WU2S-OWRT1-A-2025-5-8-14-34.tar.gz](https://github.com/user-attachments/files/20109672/supportdata-WU2S-OWRT1-A-2025-5-8-14-34.tar.gz)

aanon4, 2025-05-08:
Does either of them connect with other PtMP or Stations (ie. different hardware)?

aanon4, 2025-05-08:
I notice you have it cranked up to 80 MHz. Any difference on a more sedate 20 MHz?

wu2s, 2025-05-09:
Two tests

1) put both OpenWRT devices on channel 157 with 20 MHz bandwidth. The devices did not connect.
2) Leaving the conditions as above, put a HAP AC2 into PtMP mode and set to channel 157 with 20MHz bandwidth.
The HAP AC2 and OWRT1-B (station unit) connect successfully.

[supportdata-WU2S-OWRT1-A-2025-5-8-18-34.tar.gz](https://github.com/user-attachments/files/20113183/supportdata-WU2S-OWRT1-A-2025-5-8-18-34.tar.gz)
[supportdata-WU2S-OWRT1-B-2025-5-8-18-35.tar.gz](https://github.com/user-attachments/files/20113182/supportdata-WU2S-OWRT1-B-2025-5-8-18-35.tar.gz)

[supportdata-WU2S-OWRT1-B-2025-5-8-19-4.tar.gz](https://github.com/user-attachments/files/20113187/supportdata-WU2S-OWRT1-B-2025-5-8-19-4.tar.gz)
[supportdata-WU2S-H2-B-2025-5-8-19-5.tar.gz](https://github.com/user-attachments/files/20113188/supportdata-WU2S-H2-B-2025-5-8-19-5.tar.gz)

aanon4, 2025-05-09:
I tried the opposite test - set my One up as PtMP and had a Rocket AC as a Station - that connected.
So, to summarise so far:

* AdHoc Mode: One will not connect to anything.
* Managed mode: One will connected as a Station or AP to a non-One device, but will not connect to another One.

Let me think about this a bit in an attempt to avoid buying another One for more testing.

AI7NC, 2025-05-15:
Doesn't 'mesh ptp' mode function the same as 'mesh ptmp' except it has a MAC filter? Are you sure you set the 'mesh station' device MAC properly in the 'mesh ptp' config? 

[image]