aredn/aredn #2103: Version 3.25.5 - Port Forwards Don't Work

N6LKA, 2025-05-09:
When I upgraded to 3.25.2, it didn't copy over the port forwards. Others have already reported that issue, and re-entering the port forward information made it work fine. 

Upgrading to 3.25.5 was supposed to fix the problem so I upgraded to 3.25.5 this morning, and it did copy over the port forward list. However, none of the port forwards were working. I tried several different ways to access services on my node via a port forward, and nothing worked. I'm running my main node on a Proxmox Virtual Machine. Luckily, I have nightly backups, so I restored from my backup to version 3.25.2, and all the port forwarding is again working.

Unfortunately, I did not download the support data file before I restored from the backup, and I'm not going to reinstall the update just to get the support data file. However, I thought it was important to still report this issue as others may have the same problem.

FYI - my friend who doesn't use a virtual machine, but uses a HAP upgraded this morning as well, and he ran into the same problem where none of his port forwards worked after upgrading, even though they were listed correctly.

aanon4, 2025-05-09:
The support data before the upgrade would be a good starting point - thanks.

N6LKA, 2025-05-09:
Sure thing, Tim.  I've attached the support data from my node after restoring from my backup. I run nightly backups, so this puts my version back to 3.25.2.

[supportdata-N6LKA-PVM-QTH-2025-5-9-11-4.tar.gz](https://github.com/user-attachments/files/20126906/supportdata-N6LKA-PVM-QTH-2025-5-9-11-4.tar.gz)

aanon4, 2025-06-11:
Reopen if not fixed.

closed 2025-06-11 by aanon4