aredn/aredn #2119: WAN to LAN Port Forwarding also creates a forwarding rule for dtdlink

AI7NC, 2025-05-11:
### Describe the bug
When you create a port forwarding rule on a node for WAN to LAN, it also for some reason creates a forwarding rule on the dtdlink interface. 
This became noticeable when we tried to access our node's page routed via the dtdlink interface while having a port 80 wan to lan forward in place for a service we want accessible from the internet. Obviously we got the service page instead of the node interface which of course is undesired. 

- [ yes] Does this issue still occur on the latest version? (**Required**)
We observe this on 3.25.2.0 and 3.25.5.1 and likely much older firmware versions as well.

### Expected behavior
We expect a WAN to LAN port forward rule created in the web interface to be created for 'src wan' only, not for 'src dtdlink'.

### Support data
Attached

[supportdata-LCcore-TunSrv-2025-5-11-15-17.tar.gz](https://github.com/user-attachments/files/20151663/supportdata-LCcore-TunSrv-2025-5-11-15-17.tar.gz)

### Screenshots

[image]

[image]

### Additional context

VA2XJM, 2025-05-12:
Related to #1993

AI7NC, 2025-05-31:
Confirmed fixed on QEMU PC running 20250530 nightly.

closed 2025-05-31 by AI7NC