aredn/aredn #2130: Duplicate Neighborhood Nodes Reports with NB 20250512

K5RA, 2025-05-13:
It appears that NB 20250512 in  hAP ac Lites at KA5TYW  makes duplicate Neighborhood Nodes reports in hAP ac Lites running release 3.25.5.1 at K5RA-FB and K5RA-WH.  Reverting KA5TYW nodes to NB 20250509 fixed the problem. 

Node Status screenshots and Support Data from K5RA Nodes with KA5TYW running NB20250512.  Reboots and power-cycles of nodes at K5RA did not resolve the duplicate Neighborhood Nodes listings.

[image]

[image]

[image]

[supportdata-K5RA-HAP-FB-2025-5-12-12-17.tar.gz](https://github.com/user-attachments/files/20187244/supportdata-K5RA-HAP-FB-2025-5-12-12-17.tar.gz)

Node Status screenshots and Support Data from K5RA Nodes with KA5TYW running NB20250509.

[image]

[image]

[image]

[image]

[image]

[supportdata-K5RA-HAP-FB-2025-5-13-6-9.tar.gz](https://github.com/user-attachments/files/20187289/supportdata-K5RA-HAP-FB-2025-5-13-6-9.tar.gz)

[supportdata-K5RA-HAP-WH-2025-5-13-6-8.tar.gz](https://github.com/user-attachments/files/20187295/supportdata-K5RA-HAP-WH-2025-5-13-6-8.tar.gz)

aanon4, 2025-05-14:
https://github.com/aredn/aredn/pull/2135

K5RA, 2025-05-14:
KA5TYW upgraded three of his nodes to NB 20250514, and the issue reported here appears to have been fixed by #2135.  
Thank you very much for this quick fix!