aredn/aredn #2133: hostnames don't resolve

Orv, 2025-05-13:
### Describe the bug
Local devices are pingable by IP address but hostnames don't resolve

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes - babel-20250513-83374d73
### Expected behavior
Hostnames should be resolvable

### Support data

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-05-14:
https://github.com/aredn/aredn/pull/2134

Orv, 2025-05-14:
#2134 did not resolve issue;  hostnames still don't resolve.    Looking specifically at kj6ttn-auxcomm.

[supportdata-KJ6TTN-QTH-NODE-2025-5-14-10-30.tar.gz](https://github.com/user-attachments/files/20211922/supportdata-KJ6TTN-QTH-NODE-2025-5-14-10-30.tar.gz)

[image]

[image]

aanon4, 2025-05-14:
https://github.com/aredn/aredn/pull/2137

AI7NC, 2025-05-15:
This may be unrelated but I have observed that the node stops propagating the DNS of a device if it has been offline for some time. Without forcing a reboot it can take up to an hour after starting the device for DNS to start flowing again. I really dislike this behavior, I feel that if a hostname has been reserved it should remain in the DNS stack.

Orv, 2025-05-15:
Now works as expected.   Issue resolved.