aredn/aredn #2140: DTD connected node not showing in local node list

AI7NC, 2025-05-15:
### Describe the bug
A local DTD connected node is showing in the 'neighborhood nodes' instead of 'local nodes'. Both nodes are 3.25.5.1

- [X ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
It would expected to be in the 'local nodes' grouping instead of 'neighborhood nodes'.

### Support data

[supportdata-AI7NC-Thurston-Sec-2025-5-15-13-32.tar.gz](https://github.com/user-attachments/files/20234134/supportdata-AI7NC-Thurston-Sec-2025-5-15-13-32.tar.gz)

### Screenshots

[image]

[image]

### Additional context

aanon4, 2025-05-15:
It says it's 6 miles away. We do not consider that local regardless of how it's connected.

AI7NC, 2025-05-15:
So it's based on location not network topology?

73,

Nathan Campbell, AI7NC

K5RA, 2025-05-15:
The AREDN Docs has the following

Local Nodes "displays any local DtD nodes that are directly connected to your node, typically via Ethernet cable. "

AI7NC, 2025-05-15:
Based on that description it would lead you to think the categorization is based off the local connectivity via ethernet (DTD), not the GPS location. You can close this as a bug if it's intended to be DTD plus within a certain distance via the coordinates but the docs  should be updated to reflect that the sorting is based on location distance (with distance threshold listed), and not solely the fact that it's directly connected. 

It is common for people to spread out their node locations so you can see them on a map, my example may be extreme, but I see this issue with our hospital nodes as well (below). The node is 0.1 miles away but showing as non-local. I'm not sure what distance is right but it seems a bit tight when we have installations in 1/4 mile long buildings. 

[image]

[image]

aanon4, 2025-05-16:
https://docs.arednmesh.org/en/latest/arednGettingStarted/node_admin.html#local-nodes
DtD link must be within 100m (the max length of an ethernet cable) to be considered local.

AI7NC, 2025-05-16:
Well you got me there, we're using fiber for DTD in the Park Place building as well as Peace Health Riverbend as the runs are longer than 100m due to the size of the building.

aanon4, 2025-05-16:
If it were down to me the distance would be a lot shorter. Local is suppose to be a single place, and 100 meters away is somewhere else. But anyway ...

AI7NC, 2025-05-16:
In my mind it's all the same site. 🤷‍♂️

closed 2025-05-31 by AI7NC