aredn/aredn #2146: Tunnel Up-time Report in Pop-up

K5RA, 2025-05-16:
I woke up yesterday morning at 0300R to find that an RF link had failed between a co-located hAP ac Lite and a hAP ac2.  The tunnel pop-up on the hAP ac Lite end said link had been down for three minutes.  Five hours later when I got back to it, the pop-up reported that link was restored but there was no report for how long it had been up.  If I had not awaken at 0300 to find it, I never would have known that the tunnel had been down during the night.  

I feel it would be useful for the tunnel pop-up to display the length of time that the tunnel has been active in addition to the length of time that it has been inactive.

aanon4, 2025-05-16:
Seems like a good idea to me.

aanon4, 2025-05-16:
https://github.com/aredn/aredn/pull/2149

closed 2025-05-27 by aanon4