aredn/aredn #2148: tunnels are not attempting to reconnect

KD4WLE, 2025-05-16:
### Describe the bug
Since 3.25.5.0, tunnels are not attempting to reconnect to QEMU tunnel server node. Not sure if a bug or a misconfiguration somewhere, since this issue spans over a number of firmware revisions (Nodes connecting TO Tunnel server)

Once a node (client) is restarted, it will connect to the tunnelserver without issue.  I dont recall having this problem on older firmwares. 

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes

### Expected behavior
Nodes should attempt to reconnect after a tunnel has dropped

[supportdata-KD4WLE-Titusville-TunnelServ-2025-5-16-17-11.tar.gz](https://github.com/user-attachments/files/20257204/supportdata-KD4WLE-Titusville-TunnelServ-2025-5-16-17-11.tar.gz)

aanon4, 2025-05-16:
Could you provide the support data for both ends of a tunnel when the tunnel has disconnected but should have reconnected? Also, why are the tunnels disconnecting in the first place?

KD4WLE, 2025-05-16:
Firmware update, network outage, etc. I'll work on some support data for you, as this tunnel server has 30 clients, I sorta hate to pop an outage..

KD4WLE, 2025-05-17:
My Marion station went off last night during an internet outage at this remote location. Looks like it DID reconnect to N4TDX-Mims-Swarm, another tunnelserver on 3.25.5.1 but not the KD4WLE-Titusville-Tunnelserv. Looks like this must be an issue on my end then?

[supportdata-KD4WLE-Marion-2025-5-17-13-51.tar.gz](https://github.com/user-attachments/files/20265195/supportdata-KD4WLE-Marion-2025-5-17-13-51.tar.gz)

KD4WLE, 2025-05-25:
This was a local issue it seems. Problem can be closed.

closed 2025-05-25 by KD4WLE