aredn/aredn #2155: Tunnels from HAP showing connected but can't connect to Client GUI

k1ky, 2025-05-19:
### Describe the bug
Tunnel connections from HAP devices show 100%/100% and others with no LQ and name resolved and others either  IP addy only indicate that they are connected (UP) in the tunnel section but can't be navigated to their GUI. 
- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes 3.25.5.1

### Expected behavior
A clear and concise description of what you expected to happen.
Expect to be able to connect to tunnel clients that show as "UP"

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-W4CAT-HAP-400-CATHSE-2025-5-18-21-15.tar.gz](https://github.com/user-attachments/files/20276379/supportdata-W4CAT-HAP-400-CATHSE-2025-5-18-21-15.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

K6QS, N2MH, W4GHD and the .245 nodes cannot be navigated to via browser.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-05-19:
They look to be down. Can you show the dialog where it says they're up so I can see the details?

k1ky, 2025-05-19:
[image]

[image]

[image]

[image]

[image]

k1ky, 2025-05-19:
All of the above nodes "show" to be "UP" and are known to be operating.

aanon4, 2025-05-19:
While I appreciate the following is not a fix, we are deprecating legacy tunnels in favor of wireguard tunnels. Consequently there's not much reason to invest time into debugging something that I would advise you to replace. If the problem persists when using wireguard we should revisit.

k1ky, 2025-05-19:
Further investigation results that a computer locally connected to the node can connect to the 100%/100% nodes, but they are not accessible nor propagate through the network remotely.

k1ky, 2025-05-19:
I believe transition to Wireguard for these connections takes care of the issue.  Still working through all of them before I can fully declare success.  Will report back when migration has been completed.

k1ky, 2025-07-22:
Migrating to Wireguard. CLosing this issue.

closed 2025-07-22 by k1ky