aredn/aredn #2164: AFS doesn't find images for a RBLHGG-5acD-XL-US

Orv, 2025-05-21:
### Describe the bug
A clear and concise description of what the bug is.
AFS doesn't find images for a RBLHGG-5acD-XL-US

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes.   I verified the image is at https://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/aredn-20250521-f453a7d3-ipq40xx-mikrotik-mikrotik_lhgg-5acd-xl-squashfs-sysupgrade-v7.bin

### Expected behavior
AFS should be able to find an appropriate image for this device.

aanon4, 2025-05-21:
Could you add a link to the Mikrotik page for this device - thanks.

Orv, 2025-05-21:
If this isn't confusing, I don't know what is...
https://mikrotik.com/product/lhg_xl_5_ac

aanon4, 2025-05-21:
Well if I type it's name in I find it just fine:
https://downloads.arednmesh.org/afs/www/?version=3.25.5.1&target=ipq40xx%2Fmikrotik&id=mikrotik_lhgg-5acd-xl
I dont think AFS lets you look things up by firmware file name.

closed 2025-05-24 by aanon4