aredn/aredn #2169: Port Forwarding failing version 3.25.5.1

KA2FNK, 2025-05-22:
### Describe the bug

Using  MikroTik hAp ac3 and have migrated to release 3.25.5.1 as soon as it was released and all seemed good. 
Noticed after about a week  ( May 20) that the rsync  through my node to a raspberry pi connected to that node started to fail after being on the new release for awhile.

Checked the three other port forwarding rules and found that they were also failing. 

 I failed back to version 3.25.2.0 and all  port forwarding rules work again. Reinstall version 3.25.5.1 and the problem returned. All hardware was rebooted, but the issue persists with 3.25.5.1 with port forwarding failing.   

What is puzzling is what happen to the port forwarding after being on the 3.25.5.1 version for close to a week and 
now seeing the failure. 

[image]

[supportdata.tar.gz](https://github.com/user-attachments/files/20399929/supportdata.tar.gz)

aanon4, 2025-05-24:
Could you send me the support data for the node running 3.25.2.0 so I can compare - thanks.

KA2FNK, 2025-05-24:
I rolled the node back to 3.25.2.0. The Port forarding is working with the 3.25.2.0 version.
Here is the supportdata file from that node with 3.25.20.
Thank you Bill KA2FNK

[supportdata-3-25-25-0.tar.gz](https://github.com/user-attachments/files/20427342/supportdata-3-25-25-0.tar.gz)

aanon4, 2025-05-26:
This is odd. The firewall rules generate for both versions are identical and that's where for forwarding rule are.

aanon4, 2025-05-27:
So, if you test this path from the linux pc using `ssh -v` what does it say? Does it hang, get rejected, something else? I'd be interested in seeing the debug output.

aanon4, 2025-05-27:
I think the issue is a missing route, so while your connection into the node is fine, it cannot find its way back to you. The fix above should take care of it.

aanon4, 2025-05-27:
I think this is the final fix.
https://github.com/aredn/aredn/pull/2197

closed 2025-06-05 by aanon4