aredn/aredn #2171: 3.25.5.1 Cannot Change Channels on Rocket RM3

dbsanders, 2025-05-23:
### Describe the bug
Channel selection dropdown is populated with valid channels for selection, but selecting any of them issues a PUT request to /a/status/e/radio-and-antenna with a "null" value for radio0_channel.

- [YES] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
The option value should be the channel number, as confirmed in other models instead of null

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
[supportdata-N6IPD-IDEC-RM3-RRm-P2P-TomatoSprings-2025-5-23-11-54.tar.gz](https://github.com/user-attachments/files/20416855/supportdata-N6IPD-IDEC-RM3-RRm-P2P-TomatoSprings-2025-5-23-11-54.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
3.25.2.0 is fine, did not test 3.25.5.0

aanon4, 2025-05-23:
Could you post the same image but from 3.25.2.0 so I can compare? I dont have any 3GHz hardware so have no way to do this.

dbsanders, 2025-05-23:
Here is the same html from a Rocket M3 with 3.25.2.0.

[image]

aanon4, 2025-05-23:
I dont know if you have a way to test this fix, but if you did that would be great. I dont have hardware.
https://github.com/aredn/aredn/pull/2173

dbsanders, 2025-05-24:
That did the trick, we can now change channels!

[image]