aredn/aredn #2178: Babel Nightly 0524 issues

k1ky, 2025-05-24:
### Describe the bug
Can't connect to the internet or PBX associated with the local server.  

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes - introduced with Babel 0524, was working fine with Babel 0522.

### Expected behavior
A clear and concise description of what you expected to happen.
Expect ability to connect to the internet and have local phone register with PBS on the system.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
Attached - LHG5.  Same problems with HAP and GL750 devices

[supportdata-K1KY-LHG5-152-SMYRNA-OFFICE-COUCH-2025-5-24-18-49.tar.gz](https://github.com/user-attachments/files/20428644/supportdata-K1KY-LHG5-152-SMYRNA-OFFICE-COUCH-2025-5-24-18-49.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.
I am able to run a traceroute to 8.8.8.8 in the tools section, so there is a path to the internet available.

k1ky, 2025-05-25:
This unit connects via RF to a PTMP server node K1KY-PBELM5XW-149 which is internet enabled.  There is another node connected to the 149 unit that has the local PBX server. attached.

aanon4, 2025-05-25:
This will be related to the recent changes to how routing works. So I need a very precise description of what is connecting to what for me to debug this as all the points in the path, including exactly how they connect, matter here.

Let's just pick one failure and work with that first.

It sounds like the PBX cannot connect to the Internet - is that correct? Can you describe *exactly* how to PBX is expected to reach the Internet. Thanks.

k1ky, 2025-05-25:
OK - will work on the internet access issue first after reboot of the node.

aanon4, 2025-05-25:
I notice in your support data that you have disabled "Provide LAN devices with a default route" (lan_dhcp_defaultroute). Does your PBX have a static default route set rather than using DHCP?

k1ky, 2025-05-25:
That is disabled on all of my nodes per the documentation example.  The PBX has a static addy 10.84.136.50 so it's client phones can find it if that is what you are asking?

aanon4, 2025-05-25:
Re-reading the docs, I should still fix this so it behaves as it did before. But if you could do the test still that would be useful.

aanon4, 2025-05-25:
Actually - never mind the "Provide LAN devices with a default route" - it wont change anything. Let's get back to me understanding the PBX scenario.

k1ky, 2025-05-25:
ok - I know you won't like this part as the PBX Linux Computer is currently connected to an AirRouter (Still running olsr 3.25.5.1). The AirRouter is DTD connected to the PowerBeam Head that is currently my internet and Tunnel Server.  Currently, the Phone Devices connected through my local network and the external devices are working with 3.25.5.1.

aanon4, 2025-05-25:
Looking at the support data provided some more. The node doesnt have it's own WAN address and you have disabled the option to let LAN devices access a remote WAN over the network. So .. I'm not sure how a LAN device could access the internet from here.

This is also relevant - https://github.com/aredn/aredn/pull/2179 - as I got some of the key names wrong which would prevent you changing some options.

k1ky, 2025-05-25:
That's where we run into differing interpretations or actual performance of the settings.  What should the correct switch settings be these days for Tunnel/Internet Server and remote nodes?  To answer your question about the Computer reporting Internet is the network icon in the right bottom corner.

k1ky, 2025-05-25:
After reboot of computer and disabling local wan on the node and reboot, the computer now does not report having internet, which reflects what is happening (no Internet). Maybe await whatever "fixes" you have applied today and try again tomorrow and see what happens.

k1ky, 2025-05-26:
Nightly Babel-0525 seems to have fixed the reported issues.

closed 2025-05-26 by k1ky