aredn/aredn #2202: Connected links on Supernode are Dead

FredK27, 2025-05-28:
### Describe the bug
Recently, we have noticed that about half of the tunnel connections on our Supernode are dead. All of the links appear to be up and error-free but when they are clicked, a browser timeout error occurs.

- [ YES] Does this issue still occur on the latest version? (**Required**). We are running 3.25.1

### Expected behavior
All of the links should either work when clicked or should give an indication that they are down.

### Support data
See the screenshot of our supernode dashboard. The links to the following Supernodes work -

* hb9bla0basel-supernode
* kn6plv-bam-supernode
* m0nfs-eng-supernode
* n2mh-supernode-oki
* zl2aqy-nzi-supernode

The following links (which appear fine) are dead -

* k1vl-shrewsburyvt-supernode
* k9kdj-indianna-supernode
* ku7pdx-hio-supernode
* ve3kpg-kawarthas-supernode

All of these connections are Wireguard client  tunnels (see screenshot)

### Screenshots

[image]

[image]

### Additional context
Support data and the last portion of the Supernode's syslog are attached.

[supportdata-AB1OC-HNH-SUPERNODE-2025-5-28-19-28.tar.gz](https://github.com/user-attachments/files/20496110/supportdata-AB1OC-HNH-SUPERNODE-2025-5-28-19-28.tar.gz)

[Syslog.pdf](https://github.com/user-attachments/files/20496114/Syslog.pdf)

FredK27, 2025-05-28:
Note that we have disabled 44net per the following item in the documentation -

[image]

We have done this because some of the routers in our network have Internet connections using 44 addresses.

FredK27, 2025-05-29:
> Unfortunately clicking on them introduces the problem of where you click on them from. So let's start by using the ping tool on the supernode to check that all of them can be reached from the supernode itself. Thanks.

Hello Tim,

Thank you for working this. I can successfully ping all the problem nodes. The reverse pint (asking them to ping me) fails for all the problem nodes. All the working tunnels ping fine in both directions. Here’s an example of one of the failing connections –

***@***.***

The reverse ping just hangs –

***@***.***

Perhaps we are dealing with a 1-way tunnel connection for the problem links?

Best and 73,
Fred Kemmerer,  AB1OC

Mail: ***@***.******@***.***>
ARISS Mentor and Ground Station

FredK27, 2025-05-29:
Sorry, cut and paste did not work - here is the second screenshot -

[image]

aanon4, 2025-05-29:
I you can get on quick contact with any one of the supernode owners that is current failing for you, can you ask them to (a) get support data from their supernode and attached it here, and (b) then reboot their supernode?
Thanks.

aanon4, 2025-05-29:
[supportdata-VE3KPG-Kawarthas-Supernode-2025-5-28-21-36.tar.gz](https://github.com/user-attachments/files/20496871/supportdata-VE3KPG-Kawarthas-Supernode-2025-5-28-21-36.tar.gz)

aanon4, 2025-05-29:
[supportdata-KU7PDX-HIO-SUPERNODE-2025-5-29-7-33.tar.gz](https://github.com/user-attachments/files/20507730/supportdata-KU7PDX-HIO-SUPERNODE-2025-5-29-7-33.tar.gz)

aanon4, 2025-05-29:
Thanks for this data. Looks like processes are being killed due to a memory leak. We are tracking that here - https://github.com/aredn/aredn/issues/2192

closed 2025-05-29 by aanon4

aanon4, 2025-05-30:
The memory leak fix should fix this.

FredK27, 2025-05-31:
I'm sorry to say it, but the problem with dead links on our Supernode persists after the update to last night's build. Currently, the links between our Supernode AB1OC-HNH-SUPERDNODE and both k1vl-shrewsburyvt-supernode and ku7pdx-hio-supernode are unresponsive when clicked, even though the dashboard shows that the tunnels are up. I am attaching a screenshot and the latest support data. The connection to ku7pdx is intermittent but has mostly been unavailable for the last 30 minutes. No connectivity at all to k1vl.... An earlier ping test failed in both directions.

[image]

[supportdata-AB1OC-HNH-SUPERNODE-2025-5-30-20-37.tar.gz](https://github.com/user-attachments/files/20528433/supportdata-AB1OC-HNH-SUPERNODE-2025-5-30-20-37.tar.gz)

aanon4, 2025-05-31:
Please confirm that all the endpoints that are failing have been upgraded.

FredK27, 2025-05-31:
All of the endpoints at my site have been upgraded. I believe that all of the other supernodes that result in "dead" links from our supernode have also been upgraded to address this problem. Interestingly, the one node that I can consistently reach when other links are not working is n1mh-supernode-oki. This node is still running 3.24.10.0. Also, some of the links that are initially "dead" after a reboot of our supernode will come back if we wait 30 minutes or more, and some of these nodes are running 3.25.1.

[image]

If I reboot our supernode, all of the other supernodes mentioned as having "dead" links will have dead links. If I monitor the number of Babel nodes in the status dashboard on our supernode, the number of indicated routes fluctuates over a range from approximately 50 to sometimes approaching 500. If we wait a long time (30 minutes or more), some of the problem supernode links will start working again, with the final number of Bable nodes in the status display being around 895. The k1vl-... link does not work, no matter how long we wait. This link was working after both ends were upgraded to the nightly build on 5/30. All of the links on status display are green the entire time the dead link problem is happening.

FredK27, 2025-05-31:
There seems to be a stability problem with the Babel routes on our Supernode. After clicking on this SN connection just now -

[image]

I was able to reach the zl2aqy-... node after maybe a 30-second delay. When I came back to the status page on our supernode, the number of Babel routes in the dashboard dropped from 894 to 49 -

[image]

FredK27, 2025-05-31:
After a reboot of his Supernode (k1vl-...), the links between us are working again. A new link is down here -

[image]

Carmine appears to be experiencing the same type of dead link issues on his end that I am seeing here.

This is an interesting variation as I do not see the link to K9KDJ's node in the ping destination list on my Supernode, yet the link status is green in the dashboard. The inability to find the connection to try to ping is a new variation of the symptoms.

aanon4, 2025-05-31:
> After another 2-3 minutes, our Supernode did a self-reboot.

Curious about this as I never see any of my node, super or otherwise, spontaneously reboot. Do you capture the logs for your supernode remotely so you can see what led up to this?

> http://k9kdj-indiana-supernode.local.mesh/

They're running babel on their supernode which is why the reporting is lacking and you dont see it as a ping target as we still uses OLSR for all that data.

FredK27, 2025-05-31:
Here's the latest support data. The k9kdj-... node is currently down. The other connections are working at this time.

[supportdata-AB1OC-HNH-SUPERNODE-2025-5-31-15-50.tar.gz](https://github.com/user-attachments/files/20535686/supportdata-AB1OC-HNH-SUPERNODE-2025-5-31-15-50.tar.gz)

FredK27, 2025-05-31:
Here's what I have in my log server related to the Supernode -

[image]

The Supernode reset again somewhere between between 15:47 and 16:00

aanon4, 2025-05-31:
Thanks - just what I needed.
So it looks like k9kdj-indiana-supernode installed an early babel night build which contains a number of bugs, one of which set one of the node addresses to 10.0.0.0 which is not a good address for anything. The simplest thing to do would be to reinstall the node from scratch. Fixing it by hand it a whole bunch of steps - not impossible though.

FredK27, 2025-05-31:
So just to be clear. Are you suggesting that I reinstall our supernode AB1OC-HNH-Supernode from scratch or should K9KDJ-... be doing this? Also, do you think what you are seeing would account for the problems we are having with dead links to other supernodes?

aanon4, 2025-05-31:
Not suggesting you do anything with your supernode.

Just addressing the issue with K9KDJ's supernode. Not generalizing this to anything else. I can send them "fix it without reinstalling" instruction if you have an email for them.

FredK27, 2025-05-31:
I will send you ken's email via a personal message....

FredK27, 2025-05-31:
Additionally, since the last reboot, my tunnel to Ve3kpg's supernode has stopped working. If you look at the excerpt from my system log, you can see that I am no longer able to reach that supernode to set up the tunnel. Here's another dump of the current support data -

[supportdata-AB1OC-HNH-SUPERNODE-2025-5-31-16-51.tar.gz](https://github.com/user-attachments/files/20535877/supportdata-AB1OC-HNH-SUPERNODE-2025-5-31-16-51.tar.gz)

I also can not successfully ping his internet connection from here at this point. His down may have gone down. I am going to reboot my supernode to see if that might help. If it does, I will post here. If nothing happens in the next 10 minutes, then the reboot did not help.

iannace, 2025-05-31:
I am on 20250530-5d37834d and it appears I having intermittent issues with browsing the AB1OC tunnel. The rest of my tunnels appear to be fine with the exception of K9KDJ who has been down for at least 24 hrs and I am unsure if he is having a local issue or it is related to this issue. At Fred's request I rebooted my node this morning to help with his connectivity and my node has been up for over 9 hours since then. I've included my latest support data to this entry for reference.  

I will also build an Uptime Kuma monitor to one of Fred's working local IP's to monitor our tunnel uptime. I will use IP to avoid any potential name resolution issues.

Carmine K1VL

[supportdata-K1VL-SHREWSBURYVT-SUPERNODE-2025-5-31-16-44.tar.gz](https://github.com/user-attachments/files/20535932/supportdata-K1VL-SHREWSBURYVT-SUPERNODE-2025-5-31-16-44.tar.gz)

aanon4, 2025-06-01:
So sitting here watching the connection between our supernode and yours. Every now and again I see sudden packet loss, the RTT times spikes, and my traceroute to you fails. When the connection recovers the babel routes reported on the main page has usually dropped to a small number and then gradually increases again.  It's like babel is crashing on your end. Do you see anything in the logs?

aanon4, 2025-06-01:
So I ran the same test this evening and the link appears completely stable. Did anything change? I guess I dont know the environment your supernode is in.

FredK27, 2025-06-01:
Relative to you comment about the packet loss/RTT spikes - I'm attaching a copy of my slsog so you can take a look. I notice some errors like there -

Sat May 31 19:21:26 2025 user.notice wireguard_monitor: wgs5 endpoint supernode.ve3kpg.ca:6527 is not responding for 1748733686 seconds, trying to re-resolve hostname
Sat May 31 19:26:28 2025 user.notice wireguard_monitor: wgs5 endpoint supernode.ve3kpg.ca:6527 is not responding for 1748733988 seconds, trying to re-resolve hostname
Sat May 31 19:29:51 2025 daemon.info olsrd[5607]: Received netlink error code Invalid argument (-22)
Sat May 31 19:29:51 2025 daemon.err olsrd[5607]: . error: add route to 10.207.119.208/0.0.0.0 via 172.30.2.198 dev wgs2 onlink (Resource temporarily unavailable 11)
Sat May 31 19:31:29 2025 user.notice wireguard_monitor: wgs5 endpoint supernode.ve3kpg.ca:6527 is not responding for 1748734289 seconds, trying to re-resolve hostname
Sat May 31 19:34:38 2025 daemon.info olsrd[5607]: Received netlink error code Invalid argument (-22)
Sat May 31 19:34:38 2025 daemon.err olsrd[5607]: . error: del route to 10.207.119.208/0.0.0.0 via 172.30.2.198 dev wgs2 onlink (Resource temporarily unavailable 11)
Sat May 31 19:36:31 2025 user.notice wireguard_monitor: wgs5 endpoint supernode.ve3kpg.ca:6527 is not responding for 1748734591 seconds, trying to re-resolve hostname
Sat May 31 19:40:11 2025 daemon.info olsrd[5607]: Received netlink error code Invalid argument (-22)
Sat May 31 19:40:11 2025 daemon.err olsrd[5607]: . error: add route to 10.242.23.176/0.0.0.0 via 172.30.2.198 dev wgs2 onlink (Resource temporarily unavailable 11)
Sat May 31 19:41:32 2025 user.notice wireguard_monitor: wgs5 endpoint supernode.ve3kpg.ca:6527 is not responding for 1748734892 seconds, trying to re-resolve hostname
Sat May 31 19:44:55 2025 daemon.info olsrd[5607]: Received netlink error code Invalid argument (-22)
Sat May 31 19:44:55 2025 daemon.err olsrd[5607]: . error: del route to 10.242.23.176/0.0.0.0 via 172.30.2.198 dev wgs2 onlink (Resource temporarily unavailable 11)

These errors seem to be a consequence of the repeated (and failed) attempts to connect to the ve3kpg... supernode. It looks like the connection to this Supernode came back online sometime late last evening or early this morning. 

I am guessing the problem that we've been looking at is triggered by some sort of repeated connectivity-related problem as I can confirm that when all of the links are connected and stable (as is the case right now), the Supernode here will run for long periods of time without problems.

[AB1OC-Supernode-Log.txt](https://github.com/user-attachments/files/20540215/AB1OC-Supernode-Log.txt)

aanon4, 2025-06-01:
Thanks for the info. Here's what might be happening.
There is an "issue" (I think it's a bug, but who knows if they do) in the Linux kernel related to how they manage udp multicast sockets. UDP is unreliable, but if Linux fails to send a multicast packet out of a physical device (and we'll consider wireguard a physical device for this purpose) it will instead keep it in the socket buffer. And it will happily keep it forever, slowly filling the buffer up until nothing else will fit. There is no way to empty the socket buffer and no way to tell Linux "no, really, if you can't send the thing, please dump it .. it's unreliable after all". What standard Babel does when this happens is stop working; but we didnt think that was a good idea so instead we made it close the socket and open another one as this seems to be he only way to clear the unsent data. But detecting this takes time and data will be lost. I think, maybe, these erroring wireguard tunnels are triggering this problem and fix up. It would fit the pattern.

siddoncd, 2025-06-03:
I had a similar problem with HAP2 and HAP3 nodes.  These are Wireguard tunnel servers.  I eventually found that the WAN port was not transferring data.  If rebooted, the IP address would most often not be assigned via DHCP to the node.   I eventually went back to version 3. 23.12.  Each time I applied a newer version I noted whether the IP address was assigned via DHCP.  All worked well until version 3.25.5.1.  I had to back down to version 3.25.5.0 so that  the IP address was assigned and the WAN port would transfer data.   If you want to review the syslogs, let me know.

FredK27, 2025-06-08:
It looks like our Supernode is still periodically resetting itself. The latest rest occurred about 8 1/2 hours ago. This seems to be related to situations where one of the Supernodes we are connected to is going up and down (or at least their tunnels are). I notice this line in our syslog -

        Sat Jun  7 09:35:48 2025 user.err manager.olsrd_watchdog: olsrd watchdog timeout - restarting

I'm attaching the full syslog from our supernode as of this morning for you to look over.

[25-06-08 AB1OC Supernode Syslog.txt](https://github.com/user-attachments/files/20644691/25-06-08.AB1OC.Supernode.Syslog.txt)

FredK27, 2025-06-08:
Tool another reset about 20 minutes or so ago. Here's the syslog activity around the reset -

Sun Jun  8 16:45:44 2025 daemon.info babeld[2220]: netlink_read: increasing SO_RCVBUF to 4608K
Sun Jun  8 16:48:29 2025 user.notice wireguard_monitor: wgs5 endpoint supernode.ve3kpg.ca:6527 is not responding for 1749415709 seconds, trying to re-resolve hostname
Sun Jun  8 16:52:36 2025 daemon.info olsrd[5561]: Received netlink error code Invalid argument (-22)
Sun Jun  8 16:52:36 2025 daemon.err olsrd[5561]: . error: add route to 10.155.207.224/0.0.0.0 via 172.30.2.198 dev wgs2 onlink (Resource temporarily unavailable 11)
Sun Jun  8 16:53:30 2025 user.notice wireguard_monitor: wgs5 endpoint supernode.ve3kpg.ca:6527 is not responding for 1749416010 seconds, trying to re-resolve hostname
Sun Jun  8 16:54:34 2025 daemon.info olsrd[5561]: Received netlink error code Invalid argument (-22)
Sun Jun  8 16:54:34 2025 daemon.err olsrd[5561]: . error: del route to 10.155.207.224/0.0.0.0 via 172.30.2.198 dev wgs2 onlink (Not a tty 25)
Sun Jun  8 16:54:34 2025 daemon.err olsrd[5561]: Delete route 10.155.207.224/0 via 172.30.2.198: Not a tty
Sun Jun  8 16:54:34 2025 daemon.info olsrd[5561]: Received netlink error code Invalid argument (-22)
Sun Jun  8 16:54:34 2025 daemon.err olsrd[5561]: . error: del route to 10.155.207.224/0.0.0.0 via 172.30.2.198 dev wgs2 onlink (Resource temporarily unavailable 11)
Sun Jun  8 16:55:08 2025 daemon.info olsrd[5561]: Received netlink error code Invalid argument (-22)
Sun Jun  8 16:55:08 2025 daemon.err olsrd[5561]: . error: add route to 10.155.207.224/0.0.0.0 via 172.30.2.198 dev wgs2 onlink (Resource temporarily unavailable 11)
Sun Jun  8 16:57:20 2025 daemon.info olsrd[5561]: Received netlink error code Invalid argument (-22)
Sun Jun  8 16:57:20 2025 daemon.err olsrd[5561]: . error: del route to 10.155.207.224/0.0.0.0 via 172.30.2.198 dev wgs2 onlink (Resource temporarily unavailable 11)
Sun Jun  8 16:58:32 2025 user.notice wireguard_monitor: wgs5 endpoint supernode.ve3kpg.ca:6527 is not responding for 1749416312 seconds, trying to re-resolve hostname

As expected at this point, we have one Supernode connection that has been down since yesterday (VE3KPG). This pattern seems to be fairly repeatable.

FredK27, 2025-06-10:
Saw another spontaneous Supernode reboot this morning. They seem to be happening several times each day.

aanon4, 2025-06-10:
The OLSR watchdog is just OLSR being terrible, but doesnt reboot the supernode.
As I dont see this with my supernodes, let's start be disabling these tunnels which are not operating correctly and see if that bring stability to the node or not.

FredK27, 2025-06-10:
OK. I've disabled the one current problem connection to VE3PKG and rebooted our Supernode. All of the remaining connections appear to be working at this point. I will monitor and share any new events here.

FredK27, 2025-06-21:
After a week of testing, here is what I've found relative to our Supernode, dead links, and spontaneous resets -

1. Disabling tunnels that are sometimes dead did not stop the resets. It did slow them down somewhat. In the end, I had over 1/2 of our tunnels disabled and was still taking more than one spontaneous reset a day.
2. With all of the tunnels turned back on and the Wired Watchdog turned off, the spontaneous resets have stopped. The SN here has been up for over three and a half days with the watchdog timer turned off.
3. The logs do not indicate any meaningful data or events before or at the time of a spontaneous reset.
4. I still have some dead tunnels that show as valid in the dashboard, but they timeout when clicked on

I suspect that others are experiencing the same issues as I periodically find dead tunnels in other node dashboards.