aredn/aredn #2207: Babel Build - Prometheus - Invalid Metric Type

AI7NC, 2025-05-30:
### Describe the bug
Running babel-20250529-edde6083 on a 5AC 500 Gen2; Prometheus importer generates an error blocking metric collection. Error is 'invalid metric type "_bytes gauge".

- [x ] Does this issue still occur on the latest version? (**Required**) Only on babel nightly. Stable and regular nightly works as expected.

### Expected behavior
No error

### Support data

[supportdata-AI7NC-Thurston-PBE-2025-5-29-23-57.tar.gz](https://github.com/user-attachments/files/20516345/supportdata-AI7NC-Thurston-PBE-2025-5-29-23-57.tar.gz)

### Screenshots

[image]

### Additional context

aanon4, 2025-05-30:
https://github.com/aredn/aredn_packages/pull/120

AI7NC, 2025-06-01:
On Babel 02250531

strconv.ParseFloat: parsing "0x0": invalid syntax while parsing: "node_network_device_id{device=\"br-dtdlink\"} 0x0"

[image]

aanon4, 2025-06-01:
Good of it to tell me where the error was this time.
https://github.com/aredn/aredn_packages/pull/123

AI7NC, 2025-06-01:
I'm guessing this fix didn't make it into the 6/1 nightly? 

Have similar error.

strconv.ParseFloat: parsing "0x1003": invalid syntax while parsing: "node_network_flags{device=\"br-dtdlink\"} 0x1003"

[image]