aredn/aredn #2210: Wireless link drops after 10 minutes.

ken-mesh, 2025-05-31:
### Describe the bug
After rebooting radios a good connection is established only to be dropped after about 10 minutes.
This is a test environment of two MikroTik hAP ac3 radios.
Running babel-20250530-98f5209d

Sat May 31 06:52:08 2025 daemon.notice netifd: Interface 'wifi' has link connectivity
Sat May 31 06:52:08 2025 kern.info kernel: [  121.360772] wlan1: Selected IBSS BSSID 06:a9:2c:e6:47:9c based on configured SSID
Sat May 31 06:53:37 2025 user.notice update-time: Update clock from us.pool.ntp.org
Sat May 31 07:01:37 2025 kern.warn kernel: [  629.667950] ath10k_ahb a800000.wifi: htt tx: fixing invalid VHT TX rate code 0xff

[supportdata-W7PKA-Home-2025-5-31-7-5.tar.gz](https://github.com/user-attachments/files/20532409/supportdata-W7PKA-Home-2025-5-31-7-5.tar.gz)

- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
This has been consistent since upgrading to Babel only

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

ken-mesh, 2025-05-31:
Second radio information 

Sat May 31 00:52:07 2025 user.notice update-time: Update clock from us.pool.ntp.org
Sat May 31 00:55:09 2025 kern.warn kernel: [  331.491949] ath10k_ahb a800000.wifi: wmi: fixing invalid VHT TX rate code 0xff
Sat May 31 01:12:13 2025 daemon.info hostapd: wlan0: STA 28:f0:76:76:d0:e6 IEEE 802.11: authenticated
Sat May 31 01:12:13 2025 daemon.info hostapd: wlan0: STA 28:f0:76:76:d0:e6 IEEE 802.11: associated (aid 1)
Sat May 31 01:12:13 2025 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 28:f0:76:76:d0:e6 auth_alg=open

[supportdata-W7PKA-Attic-2025-5-31-1-14.tar.gz](https://github.com/user-attachments/files/20532428/supportdata-W7PKA-Attic-2025-5-31-1-14.tar.gz)

aanon4, 2025-05-31:
I'd be interested if the connection is more stable with the bandwidth set to 20 MHz.
The issues looks to be down in the radio somewhere and not something I have a lot of insight into.

ken-mesh, 2025-05-31:
Update to babel-20250531-ad138fca no change.
Switch from 10MHz to 20MHz, no change
Link still disconnects between 8 and 10 minutes. 
I should note that this was stable with 3.24.10.0

aanon4, 2025-05-31:
If you spend any time looking at log files for radio devices (and I unfortunately do) then you'll see stuff like this all the time. It is not, on its own, significant I'm afraid. Does the timing line up exactly when connectivity fails?
A couple more things to try (both in 20MHz):
1. What is the reliability like if you use PtMP + Station modes?
2. When connectivity fails, what does a 'wifi scan' tell you, and does connectivity get restored after you run it?

ken-mesh, 2025-06-01:
[image]

aanon4, 2025-06-01:
If you're up for it, I'll send you a test build tomorrow for the AC3 which will tweak how the radio is setup on this specific device.

ken-mesh, 2025-06-01:
I have setup http://w7pka-attic.local.mesh/ as PTMP
and setup http://w7pka-home.local.mesh/ as Mesh Station

And the link seems solid.

ken-mesh, 2025-06-02:
There is something different about the MESH setting compared to PTMP. The PTMP link has been up for 17 hours.

[image]

aanon4, 2025-06-02:
Yes there is - it's one of the reason we introduced the PtXP modes. Mesh mode uses wifi's adhoc standard which is very much the redheaded stepchild of the wifi world whom no one loves. Support for it can be rough and temperamental especially on newer radios. For new setups I'd use the PtXP modes as they're better supported in the low level firmware (which is just a blob to us) and tend to perform a little better.