aredn/aredn #2215: Rocket M3 nodes refusing to connect since NB Babel 20250515

k6ccc, 2025-06-01:
First a little background.  I have three Rocket M3 nodes at my house.  They have been happily connected for several years (not really doing anything, but have some future use).  Since they are not doing anything at the moment, they get almost every nightly build.  When Babel was introduced to the nightly builds, they started operating using Babel.  When PtP and PtMP were added, these three nodes were converted to a PtMP / Mesh Station arrangement.  When the first Babel only nightly was released, they were converted to Babel only.  Been working fine in PtMP and Babel.

K6CCC-RocM3-1 (Mesh station) is DtD connected to my hAPac3, and is using dummy loads for antennas.
K6CCC-RocM3-3 (PtMP) is about 2 feet from RocM3-1 and using quarter wave whips for antennas.
K6CCC-Johnstone-P2P-Pleasants (Mesh Station) is about 40 feet from RocM3-3 and using the dish feed only from a RocketDish.  It is DtD connected to several other nodes that will eventually be at Johnstone.
I have LAN and an alternate RF route to the Johnstone node, and DtD connectivity to RocM3-1.  Normally RocM3-3 is RF connected only.
Attached is a drawing showing the arrangement.

### Describe the bug
When nightly build 20250515 was released I updated all there nodes remotely (from work).  RocM3-1 and the Johnstone node came up fine, and connected to the DtD devices as expected, however the RF link to RocM3-3 did not connect.
Because of a insane schedule I did not have time at home to do anything about this until a couple days ago.
I connected a laptop to RocM3-3 and was surprised to find that the update to NB 20250515 had worked and the node was running fine as a stand alone node.  I updated all three 3 GHz nodes to NB 20250530 and that did not help the connectivity.  After letting the three nodes sit for an hour I changed all three to Mesh mode rather than the PtMP / Mesh Station.  That also did not help.  All three stations are showing no signal on a WiFi Signal and oddly enough on a WiFi Scan are reporting seeing various nearby (withing a few feet) 5 GHz nodes.  Screen captures of the WiFi Scan and WiFi Signals are attached.  Support data files are also attached.
This morning, all three nodes were updated to NB 20250601 with no changes.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes -

[supportdata-K6CCC-RocM3-3-2025-6-1-8-9.tar.gz](https://github.com/user-attachments/files/20541419/supportdata-K6CCC-RocM3-3-2025-6-1-8-9.tar.gz)
[supportdata-K6CCC-RocM3-1-2025-6-1-8-54.tar.gz](https://github.com/user-attachments/files/20541420/supportdata-K6CCC-RocM3-1-2025-6-1-8-54.tar.gz)
[supportdata-K6CCC-Johnstone-P2P-Pleasants-2025-6-1-8-53.tar.gz](https://github.com/user-attachments/files/20541421/supportdata-K6CCC-Johnstone-P2P-Pleasants-2025-6-1-8-53.tar.gz)

[image]

[image]
[image]

[image]
[image]
[image]

 NB 20250601

### Expected behavior
The RF should connect

### Support data
Attached

### Screenshots
Attached

### Additional context
That should be enough....

aanon4, 2025-06-01:
For some reason your nodes dont have an IPv6 link local address on their wifi interfaces. This is something babel uses to communicate.  There is an appropriate address on every other interface, so I'm not sure why this one is different. I dont have any M3 hardware so I can't replicate this setup.

Can you reboot one of your M3 nodes and then immediately get the support data? The logs attached are just full of babel whining about the lack of IPv6 so if there were earlier errors they've been lost.

Thanks

k6ccc, 2025-06-02:
Here you go.  I actually am giving you two of them...

[supportdata-K6CCC-RocM3-3-2025-6-1-19-33.tar.gz](https://github.com/user-attachments/files/20544629/supportdata-K6CCC-RocM3-3-2025-6-1-19-33.tar.gz)
[supportdata-K6CCC-Johnstone-P2P-Pleasants-2025-6-1-20-28.tar.gz](https://github.com/user-attachments/files/20544630/supportdata-K6CCC-Johnstone-P2P-Pleasants-2025-6-1-20-28.tar.gz)

aanon4, 2025-06-02:
Thanks.
By any chance to you have an M3 in service and operational with a wireless link? I'd like to compare its support data. The kernel is claiming "unknown channel" but the value looks fine as far as I'm aware, but I dont have anyway to compare it.

k6ccc, 2025-06-02:
No.  That is all three of my M3 Rockets.

aanon4, 2025-06-03:
Yes. Available tomorrow morning.

k6ccc, 2025-06-03:
Very good.  I will test tomorrow.

Thanks

k6ccc, 2025-06-03:
Loaded Babel-20250603 onto all three of my Rocket M3 nodes.  They connected right up in standard Mesh mode.  After they were stable for a few minutes, changed RocM3-3 to PtMP and RocM3-1 and Johnstone to Mesh Station.  Again, they connected right up with each other.  One question related to that.  When operating in PtP or PtMP, is the requirement to specify the MAC of the connecting Mesh Station no longer there - I could not find anyplace to enter it.

aanon4, 2025-06-03:
The station peer mac is only there if you select PtP mode, and I see the field if I do that. Where are you looking?

aanon4, 2025-06-03:
Anyway, closing this as fixed.

closed 2025-06-03 by aanon4