aredn/aredn #2233: chatty logs with ipv6 messages

komacke, 2025-06-05:
### Describe the bug
The aredn logs are getting 3 lines every 25 seconds. Here's a snippet:

Thu Jun  5 16:34:56 2025 daemon.info babeld[1932]: Warning: max_rtt_penalty is set but timestamps are disabled on interface br-dtdlink.
Thu Jun  5 16:34:56 2025 daemon.info babeld[1932]: Interface br-dtdlink has no link-local address.
Thu Jun  5 16:34:56 2025 daemon.info babeld[1932]: setsockopt(IPV6_LEAVE_GROUP): Address not available

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes, I'm on version: babel-20250605-da701c09

### Expected behavior
Much less chatty.

### Support data
[supportdata-N9DK-node1-2025-6-5-16-36.tar.gz](https://github.com/user-attachments/files/20619583/supportdata-N9DK-node1-2025-6-5-16-36.tar.gz)

### Screenshots
From "logread -f" at the command line as pasted above.

### Additional context
LMK if I can provide anything about my network.

aanon4, 2025-06-05:
Looks like this is a side effect of you having no physical ethernet connected to the node. Functionally it's fine but unnecessarily chatty about this. As I'm not going to disable these messages, I need to think of a way to make babel happy which this particular configuration. Stay tuned.

aanon4, 2025-06-06:
https://github.com/aredn/aredn/pull/2234