aredn/aredn #2241: ERROR: left-hand side expression is null

k4spblife, 2025-06-09:
### Describe the bug
A clear and concise description of what the bug is.

On my supernode running nightly build babel-20250605-da701c09 I get the below message when trying to configure tunnels:

ERROR: left-hand side expression is null
In main(), file /app/main/status/e/tunnels.ut, line 116, byte 20:
  called from function call ([C])
  called from function render ([C])
  called from anonymous function (/app/root.ut:614:18)

 `    available[`${w[0]}.${w[1]}.${w[2]}.${w[3]}:${p}`] = 1;`
  Near here ---------^

Note: issue does not occur on regular build, only Babel

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes. I updated to nightly build 20250607 and the issue persists

### Expected behavior
A clear and concise description of what you expected to happen.

Tunnel configuration bubble pops up as normal.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-K4SPB-MEMPHIS-SUPERNODE-2025-6-6-4-46.tar.gz](https://github.com/user-attachments/files/20651023/supportdata-K4SPB-MEMPHIS-SUPERNODE-2025-6-6-4-46.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.
[image]

### Additional context
Add any other context about the problem here.
First reported here:
https://www.arednmesh.org/content/error-left-hand-side-expression-null

aanon4, 2025-06-09:
So because we removed the older vtun in babel-only release, we migrated some properties and removed unused ones. The odd thing here is that the new property isn't being setup and I'm not yet sure why not.

If I send you a script can you load it onto your supernode, run it, and send me the output?

aanon4, 2025-06-09:
Also, you says:
```Note: issue does not occur on regular build, only Babel```
You cannot switch between babel and non babel builds.  Hopefully you didnt as it will break things.

closed 2025-07-02 by aanon4