aredn/aredn #2242: DRM Repeater can not connect to Maser (Branmister) on Nightly when utllizing RF connections.

KD4WLE, 2025-06-09:
### Describe the bug

Issue - DRM Repeater can not connect to Maser (Branmister) 
Repeater fails to connect to master only when running a recent nightly. The repeater will connect if node is running 3.25.5.1 downstream RF nodes do not impact connectivity on latest stable or nightly.  (n4tdx-parish-gw.local.mesh)

This issue **IS NOT** present if the Mikrotik’s WAN is directly connected to the internet with the node on the latest nightly. (n4tdx-edgewater-gw.local.mesh)

Equipment – 
Repeater – Hytera RD982

AREDN Node (Directly connected to repeater)  Mikrotik hAP ac lite at both sites.

- [Yes] Does this issue still occur on the latest version? (**Required**)

### Expected behavior

Repeater should be able to connect to master

### Support data

[supportdata-N4TDX-Edgewater-GW-2025-6-9-14-35.tar.gz](https://github.com/user-attachments/files/20656522/supportdata-N4TDX-Edgewater-GW-2025-6-9-14-35.tar.gz)
[supportdata-N4TDX-Parish-GW-2025-6-9-14-34.tar.gz](https://github.com/user-attachments/files/20656523/supportdata-N4TDX-Parish-GW-2025-6-9-14-34.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context

This same issue is being reported by Chuck KP4DJT on Slack with different repeater type. We both have common issue when using an RF link vs a node directly connected to the internet.

aanon4, 2025-06-09:
Could you describe your setup a little more? I see two support data files but I'm not sure how they related to each other, where the internet connection in this setup is, or where your repeater is connected.

Also, and probably unrelated, but both log files are full of lines like this:

```
Mon Jun  9 13:58:38 2025 daemon.err uhttpd[1746]: gzip: write error: Broken pipe
Mon Jun  9 14:00:38 2025 daemon.err uhttpd[1746]: gzip: write error: Broken pipe
Mon Jun  9 14:04:38 2025 daemon.err uhttpd[1746]: gzip: write error: Broken pipe
Mon Jun  9 14:14:31 2025 daemon.err uhttpd[1746]: gzip: write error: Broken pipe
Mon Jun  9 14:14:38 2025 daemon.err uhttpd[1746]: gzip: write error: Broken pipe
Mon Jun  9 14:16:31 2025 daemon.err uhttpd[1746]: gzip: write error: Broken pipe
```

Any idea what this would be about?

KD4WLE, 2025-06-09:
The node with the issue is n4tdx-parish-gw.local.mesh, this device connects to N4TDX-Parish-MVFD-PTP via DTD and then an RF link to n4tdx-mims-mvfd222-ptp and a DTD to n4tdx-parish-gw and so forth until it reaches it's internet gateway. 

n4tdx-parish-gw.local.mesh is running 3.25.5.1, everything else in it's path is on a nightly. 20250607-010b358d or newer.

The edgewater support file is a node that has a direct WAN connection to an internet path. There is NO RF links involved. 

There is an external tool (Zabbix) that will ask for sysinfo and metrics, along with a PING. I don't know where gzip is involved from these, but, i'm sure it is causing the broken pipe issue. Ive never noticed the issue.

aanon4, 2025-06-09:
So the Repeater is attached to n4tdx-parish-gw as a LAN device?
What have you discovery when debugging this so far? Were you able to ping the Internet and Branmister from the suspect node? What does the traceroute look like?

KD4WLE, 2025-06-09:
Yes, the Repeater is attached to n4tdx-parish-gw as a LAN device

I can ping the repeater itself from remote nodes. I can successfully ping and traceroute the master Branmister server. We have tried different masters, same result. 

Traceroute with firmware 3.25.5.1, repeater is connected (Hops 4-5, even though #4 has internet conectivity, I have had issues with that provider disrupting my connection to Branmister, so we tunnel to the KD4WLE node on AT&T) - 

root@N4TDX-Parish-GW:~# traceroute 3102.repeater.net
traceroute to 3102.repeater.net (74.91.114.19), 30 hops max, 46 byte packets
 1  dtdlink.N4TDX-Parish-MVFD-PTP.local.mesh (10.181.32.11)  1.120 ms  0.597 ms  0.580 ms
 2  N4TDX-Mims-MVFD222-PTP.local.mesh (10.172.154.182)  3.864 ms  6.212 ms  15.285 ms
 3  dtdlink.N4TDX-Mims-MVFD222-GW.local.mesh (10.72.242.27)  12.702 ms  21.349 ms  4.731 ms
 4  mid11.N4TDX-Mims-Swarm (172.31.128.70)  21.338 ms  10.538 ms  6.331 ms
 5  mid5.KD4WLE-Titusville-TunnelServ (172.31.234.132)  43.522 ms  42.511 ms  38.914 ms
 6  *  192.168.80.1 (192.168.80.1)  39.869 ms  *
 7  192.168.1.254 (192.168.1.254)  37.968 ms  *  37.241 ms
 8  108-94-100-1.lightspeed.dybhfl.sbcglobal.net (108.94.100.1)  39.498 ms  37.810 ms  39.384 ms
 9  99.168.25.54 (99.168.25.54)  44.569 ms  57.194 ms  55.382 ms
10  *  *  *
11  *  *  *
12  *  *  *
13  *  *  *
14  ae11.cr7-dal3.ip4.gtt.net (213.200.115.26)  68.731 ms  74.775 ms  73.531 ms
15  ip4.gtt.net (208.97.215.122)  81.621 ms  80.728 ms  73.806 ms
16  3102.master.brandmeister.network (74.91.114.19)  73.225 ms  85.908 ms  77.224 ms
root@N4TDX-Parish-GW:~# 

Traceroute with current nightly, repeater is offline. 

root@N4TDX-Parish-GW:~# traceroute 3102.repeater.net
traceroute to 3102.repeater.net (74.91.114.19), 30 hops max, 46 byte packets
 1  dtdlink.N4TDX-Parish-MVFD-PTP.local.mesh (10.181.32.11)  0.602 ms  0.589 ms  0.582 ms
 2  N4TDX-Mims-MVFD222-PTP.local.mesh (10.172.154.182)  5.273 ms  22.343 ms  2.102 ms
 3  dtdlink.N4TDX-Mims-MVFD222-GW.local.mesh (10.72.242.27)  5.093 ms  5.238 ms  7.336 ms
 4  mid11.N4TDX-Mims-Swarm (172.31.128.70)  9.372 ms  4.941 ms  5.262 ms
 5  mid5.KD4WLE-Titusville-TunnelServ (172.31.234.132)  44.828 ms  46.463 ms  37.675 ms
 6  192.168.80.1 (192.168.80.1)  36.280 ms  *  *
 7  192.168.1.254 (192.168.1.254)  44.396 ms  *  40.032 ms
 8  108-94-100-1.lightspeed.dybhfl.sbcglobal.net (108.94.100.1)  39.001 ms  52.242 ms  41.179 ms
 9  99.168.25.54 (99.168.25.54)  48.368 ms  41.458 ms  68.166 ms
10  *  *  *
11  *  *  *
12  *  *  *
13  192.205.32.90 (192.205.32.90)  98.779 ms  *  *
14  ae11.cr7-dal3.ip4.gtt.net (213.200.115.26)  67.712 ms  67.507 ms  75.963 ms
15  ip4.gtt.net (208.97.215.122)  73.783 ms  84.687 ms  75.349 ms
16  3102.master.brandmeister.network (74.91.114.19)  77.059 ms  80.487 ms  84.774 ms
root@N4TDX-Parish-GW:~# 

Traceroute TO repeater from remote node.

seanhaga@MacBook-Pro-5 ~ % traceroute  10.122.41.142
traceroute to 10.122.41.142 (10.122.41.142), 64 hops max, 40 byte packets
 1  10.239.185.235 (10.239.185.235)  5.754 ms  4.780 ms  5.326 ms
 2  localnode.local.mesh (10.239.185.225)  4.501 ms  6.566 ms  3.669 ms
 3  mid4.n4tdx-mims-swarm (172.31.128.90)  36.798 ms  36.613 ms  39.305 ms
 4  mid1.n4tdx-mims-mvfd222-gw (172.31.128.71)  42.093 ms  37.506 ms  32.593 ms
 5  dtdlink.n4tdx-mims-mvfd222-ptp.local.mesh (10.173.154.182)  35.442 ms  39.567 ms  52.218 ms
 6  n4tdx-parish-mvfd-ptp.local.mesh (10.180.32.11)  44.178 ms  41.990 ms  42.556 ms
 7  dtdlink.n4tdx-parish-gw.local.mesh (10.235.209.71)  43.092 ms  43.908 ms  70.258 ms
 8  10.122.41.142 (10.122.41.142)  42.980 ms  43.799 ms  62.932 ms
seanhaga@MacBook-Pro-5 ~ %

kp4djt, 2025-06-10:
Sean indicated that his problem link would work if the host router node was  3.25.5.1, I converted that node from Babel nightly to 3.25.5.1 and the repeater immediately came back on line.

aanon4, 2025-06-10:
I'll try to take a look at this problem today.
I should note that we do not support reverting a babel only node to release. Some changes are not backwards compatible and things will break. This was documented in the original release nodes when we made the babel nightly available.

kp4djt, 2025-06-10:
> The consensus view on this has been that things plugged into your LAN
> shouldn't just have access to the Internet unless you purposely enable it -
> so it's off by default.

Re-installed Babel nightly and made sure the LAN to MESH WAN switch was
green. Kerchunked the repeater
on the Parrot, 2nd kerchunk the parrot came back at me with my audio so it
is fixed.

closed 2025-06-11 by aanon4