aredn/aredn #2248: Node is showing a neighbor that isn't a neighbor

AI7NC, 2025-06-12:
### Describe the bug
KD1HA-BUR shows as a neighbor on a tunnel that is disconnected from the client side. Client is not KD1HA-BUR

- [X] Does this issue still occur on the latest version? (**Required**) | Error observed running 5/30 nightly upgraded to 6/12 and waiting to see if it shows back up or not (happens after a period of time. 

### Expected behavior
No neighbor shown for that tunnel as it's disabled on the client side. 

### Support data

[supportdata-LCcore-Main-2025-6-12-12-18.tar.gz](https://github.com/user-attachments/files/20715040/supportdata-LCcore-Main-2025-6-12-12-18.tar.gz)

### Screenshots

[image]

[image]

[image]

### Additional context
Add any other context about the problem here.

AI7NC, 2025-06-12:
I should note that from LCcore-Main's perspectvie, KD1HA-BUR is only accessible via a supernode. 

[image]

AI7NC, 2025-06-13:
The issue was reproduceable with current 6/12 nightly after cycling the tunnel on/off from the client side.

Relevant log entries: 

Fri Jun 13 10:16:50 2025 daemon.info babeld[2029]: Received prefix with no router id.
Fri Jun 13 10:16:50 2025 daemon.info babeld[2029]: Couldn't parse packet (8, 14) from fe80::200:acff:fe1f:28b9 on wgc0.
Fri Jun 13 10:16:50 2025 daemon.info babeld[2029]: Received prefix with no router id.
Fri Jun 13 10:16:50 2025 daemon.info babeld[2029]: Couldn't parse packet (8, 14) from fe80::200:acff:fe1f:28b9 on wgc0.
Fri Jun 13 10:16:50 2025 daemon.info babeld[2029]: Received prefix with no router id.
Fri Jun 13 10:16:50 2025 daemon.info babeld[2029]: Couldn't parse packet (8, 14) from fe80::200:acff:fe1f:28b9 on wgc0.
Fri Jun 13 10:16:50 2025 daemon.info babeld[2029]: Received prefix with no router id.
Fri Jun 13 10:16:50 2025 daemon.info babeld[2029]: Couldn't parse packet (8, 14) from fe80::200:acff:fe1f:28b9 on wgc0.

[image]

[supportdata-LCcore-Main-2025-6-13-10-51.tar.gz](https://github.com/user-attachments/files/20730823/supportdata-LCcore-Main-2025-6-13-10-51.tar.gz)

[image]

aanon4, 2025-06-17:
Did you work out what node was on the other end of wgc0 and why it was sending you malformed babel routing packets?

aanon4, 2025-06-17:
My **guess** then would be that some other tunnel out there associated with the unexpected hostname uses this ip address also. It's not a surprise really given the limited address space we use for them. Babel will no longer make these 172.X addresses resolvable, treating them as link local and without names, so this issue will go away then.