aredn/aredn #2253: Low memory issues on 64MB PBE 400AC

AI7NC, 2025-06-18:
### Describe the bug
After ~4 days of uptime the device runs low on memory driving CPU utilization to 300%+ which causes periodic connectivity and added latency/jitter. 

- [ ] Does this issue still occur on the latest version? (**Required**) On nightly 5/30. Will upgrade to current nightly and wait 4-5 days to replicate but same behavior is expected. 

### Expected behavior
Stable behavior. 

### Support data

[supportdata-AI7NC-Thurston-PBE-2025-6-17-20-28.tar.gz](https://github.com/user-attachments/files/20788188/supportdata-AI7NC-Thurston-PBE-2025-6-17-20-28.tar.gz)

### Screenshots

[image]

[image]

[image]

[image]

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
I'll leave the node as is for a while as nothing is depending on it to see if it ultimately self-resolves. I'll also await feedback before upgrading to current nightly to see if the condition is repeatable with a later version.

AI7NC, 2025-06-18:
Example of high CPU utilization: 

[image]

aanon4, 2025-06-18:
Could you put something a little more current on it and repeat your experiment? Thanks.

aanon4, 2025-06-18:
I'm going to revisit disabling wpa_supplicant for STAs to save some memory.

aanon4, 2025-06-18:
https://github.com/aredn/aredn/pull/2256
Let's see if this helps you. 
I noticed that we also have a 64MB PBE 400AC. It's in PtMP mode but hasn't given me any issue.

AI7NC, 2025-06-19:
I'll give that a shot once released.

On your ac are you monitoring CPU/memory/ connectivity?

Nathan Campbell

AI7NC, 2025-06-20:
Doing ok so far, there was a few missed Prometheus pulls during the first hour of being on line but it then stabilized and hasn't had any serious connectivity issues yet. Will continue to monitor to see if this solves the longer term stability issue.

[image]

aanon4, 2025-06-21:
As some issues were noted with the first attempt at this, please try tomorrow's (Satuday) nightly to make sure things are still good. Thanks.

AI7NC, 2025-06-21:
I will upgrade it tomorrow. So far I haven't had the RF disconnection issue the other individual had. Blanton-Sec is RF connected via the Thurston-PBE. It has been stable for ~18 hours. PBE is running PTMP and the sec is running mesh station.

[image]

aanon4, 2025-06-21:
I broke the station end. It needs the supplicant but not hostapd.

AI7NC, 2025-06-21:
Ah ok, The node running mesh station is on older firmware still, That's
probably why I didn't see it.

Nathan Campbell

AI7NC, 2025-06-27:
We have been stable for 6 days. The one break in the data marked was due to an upstream node reboot and isn't indicative of instability on the PBE. 

[image]

Here is the sector data. Access to the sector is via the PBE over local RF demonstrating the PBE is stably passing traffic.

[image]

AI7NC, 2025-07-09:
The 6/21 nightly has been running successfully for 18 days without any high CPU load stemming from low memory issues. Memory appears stable without a serious deficit as before. 

[image]

closed 2025-07-09 by AI7NC