aredn/aredn #2275: Node Does Not Appear on AREDN World Map

K5RA, 2025-06-26:
Why does my HAP-TC (a hAP ac lite - TC)  node not appear on the AREDN World Map?  It has been up for at least five days.  It shows my six nodes that are on the map as Local Nodes, and it shows as a Local Node to them.   Its WAN port is connected to ISP through a switch, its DtD port is attached to a DtD port of my HAP31 (a hAP ac3).  Its stated coordinates put it near the middle of the cluster of my other nodes on the Map.  Its Internal Services and Network Advanced Options appear to be same as those of my HAP-FB.  

HAP-TC is running a Babel NB, as is AHM2 and the other HaLow nodes on the Map.  Non-HaLow nodes are running recent OLSR/Babel NB.

[supportdata-K5RA-HAP-TC-2025-6-26-6-34.tar.gz](https://github.com/user-attachments/files/20927116/supportdata-K5RA-HAP-TC-2025-6-26-6-34.tar.gz)

[image]

[image]

aanon4, 2025-06-27:
When I quickly check this yesterday, the world dns could not resolve this node name. It appears there is not a babel route from your supernode to this node; not because the nodes aren't running babel builds, but that babel has crashed on one of them (probably the 3.25.5.1 memory leak).

K5RA, 2025-06-27:
I appreciate your looking into this.  

I have upgraded five HAP nodes to latest NBs.  My HAP-FB has a link to the gateway of the NTx Supernode (KE5YZP).  The gateway and the SN both running 3.25.5.1, and have been up for almost 44 days.  Our mesh island has a link to the STx SN (N5MXI).  The gateway and SN both running 3.25.5.1.  I can hop to the STx gateway (been up for almost 6 days), but nothing beyond it.  I can hop to the NTx SN, but nothing beyond it.  

I will e-mail Miranda asking for reboots, and KA5TYW will e-mail Dave about his.  Hopefully at least one SN link will be available for the 2200 Map update.

K5RA, 2025-06-27:
My five HaLow nodes disappeared from the 1623R World Map.  My MM-HL11 MorseMicro node bricked on an upgrade a few days ago, and I retired it.  My AHM12 HaLow is connected to my DtD bus, but it is Babel-only, so is no more visible than my HAP-TC.

closed 2025-06-29 by K5RA