aredn/aredn #2280: Tons of ping attempt failures in syslog even though node's repository URL was changed.

KenArck, 2025-06-27:
### Describe the bug
Even though firmware repository is changed to one on our local mesh, node continues to ping downloads.arednmesh.org. This necessarily eats up syslog (is it stored in RAM?)

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] yes. 

### Expected behavior
Ping should either be changed or stopped after x amount of tries

### Support data

[supportdata-AH6LE-CHEHALEM-5AC-PTP-2025-6-27-15-25.tar.gz](https://github.com/user-attachments/files/20957069/supportdata-AH6LE-CHEHALEM-5AC-PTP-2025-6-27-15-25.tar.gz)

[image]

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-06-27:
There are various subsystems this can come from (and I will fix them all), but did you also update the packages location? There is also a messages location but currently this is not configurable.

aanon4, 2025-06-27:
https://github.com/aredn/aredn/pull/2281

KenArck, 2025-06-28:
Changes the packages' download URL doesn't stop the log entries.

aanon4, 2025-06-28:
There we also fixes yesterday (see the attached PRs above). Did you test those also?

KenArck, 2025-07-05:
Sorry for the delayed response. Been running current nightlies on this node for a week.  Syslog screencap attached

[image]

aanon4, 2025-07-05:
That's the one that pings the aredn notifications server. Currently no way to disable it. See my https://github.com/aredn/aredn/issues/2280#issuecomment-3014601080 about this earlier.