aredn/aredn #2302: Rocket M3 radio channel selection only shows frequency

k6ccc, 2025-07-03:
### Describe the bug
On the Radios & Antennas setup page on a Rocket M3, only the frequency is shown.  The channel number does not show.  Confirmed on 5 GHz Rocket nodes that the frequency and channel number displays.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes - NB 20250703

### Expected behavior
Both frequency and channel number should display as it does on other types of nodes.

### Support data

[supportdata-K6CCC-RocM3-1-2025-7-3-9-51.tar.gz](https://github.com/user-attachments/files/21041739/supportdata-K6CCC-RocM3-1-2025-7-3-9-51.tar.gz)

### Screenshots

[image]

[image]

One of my 5 GHz Rockets shows for comparison only
[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-07-03:
I will need access to an M3 radio to fix this.

k6ccc, 2025-07-03:
Physical or remote access?  I can do some router magic to get you remote access.  
Note that I am getting ready for my 4th of July fireworks show so may be slow on responses until Saturday.

aanon4, 2025-07-03:
Remote is all I need.
And no rush - this change wont make it in this side of a release anyway as it's cosmetic.