aredn/aredn #2324: Setting channel on 900 MHz node shuts down RF

pmilazzo, 2025-07-12:
### Describe the bug
On 900 MHz nodes (both Rocket M9 and NanoStation Loco M9) running 3.25.5.1 and later, setting the channel / frequency results in:
- no RF output (confirmed with spectrum analyzer)
- iwinfo reports "Channel: unknown"
- Radio section on information page displays nonsense

A 900 MHz node in which the frequency was only set with the old UI and then upgraded to 3.25.5.1 or later will operate correctly and show the corresponding 2.4 GHz channel number, although the frequency display is empty. After subsequently changing the frequency through the new UI, however, the RF shuts down and the channel number becomes nonsense (e.g. -299), but the frequency display shows the expected value.

If the node is then downgraded to 3.25.2.0, the frequency can be reset and the node will operate correctly. Nevertheless, if subsequently re-upgraded to 3.25.5.1, it will revert to the settings previously established under that version and will not operate correctly.

- [x] Does this issue still occur on the latest version? Yes, as of 20250707-307d9e2f

### Expected behavior
After changing the frequency setting, the node should display the new channel and frequency and transmit on that frequency.

### Support data

[supportdata-K3PGM-M900-Field-2025-7-12-18-9.tar.gz](https://github.com/user-attachments/files/21198703/supportdata-K3PGM-M900-Field-2025-7-12-18-9.tar.gz)
[supportdata-K3PGM-M900b-2025-7-12-18-8.tar.gz](https://github.com/user-attachments/files/21198704/supportdata-K3PGM-M900b-2025-7-12-18-8.tar.gz)

[The support data for W3DRA-EB-9-SE is not currently accessible; sorry.]

### Screenshots
#### Rocket M9 running 3.25.2.0:

[image]

```root@W3DRA-EB-9-SE:~# iwinfo
wlan0     ESSID: "AREDN-5-v3"
          Access Point: 68:72:51:68:8A:59
          Mode: Ad-Hoc  Channel: 4 (2.427 GHz)  HT Mode: HT20
          Center Channel 1: 4 2: -1
          Tx-Power: 28 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -95 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 168C:002A 0777:E1B9 [Ubiquiti Rocket M9]
          TX power offset: 6 dB
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```
#### NS loco M9 running 3.25.2.0:

[image]

```root@K3PGM-M900b:~# iwinfo
wlan0     ESSID: "AREDN-5-v3"
          Access Point: 68:72:51:6A:F9:C3
          Mode: Ad-Hoc  Channel: 5 (2.432 GHz)  HT Mode: HT20
          Center Channel 1: 5 2: -1
          Tx-Power: 28 dBm  Link Quality: 70/70
          Signal: -30 dBm  Noise: -95 dBm
          Bit Rate: 130.0 MBit/s
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 168C:002A 0777:E009 [Ubiquiti NanoStation Loco M9]
          TX power offset: 6 dB
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```

#### Rocket M9 running 20250707-307d9e2f but frequency was previously set with old UI:

[image]

```root@K3PGM-M900-Field:~# iwinfo
wlan0     ESSID: "AREDN-5-v3"
          Access Point: 68:72:51:68:89:65
          Mode: Ad-Hoc  Channel: 5 (2.432 GHz)  HT Mode: HT20
          Center Channel 1: 5 2: -1
          Tx-Power: 28 dBm  Link Quality: 70/70
          Signal: -26 dBm  Noise: -95 dBm
          Bit Rate: 130.0 MBit/s
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 168C:002A 0777:E1B9 [Ubiquiti Rocket M9]
          TX power offset: 6 dB
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```

#### Aforementioned NS loco M9 after upgrading to 3.25.5.1:
Note that the Frequencies display has mysteriously shifted to 915 - 919 MHz, the channel number is -298, and iwinfo shows Channel: unknown:

[image]

```root@K3PGM-M900b:~# iwinfo
wlan0     ESSID: unknown
          Access Point: 68:72:51:6A:F9:C3
          Mode: Ad-Hoc  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: -1 2: -1
          Tx-Power: 28 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 168C:002A 0777:E009 [Ubiquiti NanoStation Loco M9]
          TX power offset: 6 dB
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```
Note: the frequency of this node was set to 917 MHz previously while running 20250707-307d9e2f, which I suspect is why it reverted to that frequency after upgrading. Subsequently setting the frequency to 912 MHz changes the display to:

[image]

...but iwinfo still shows "Channel: unknown" and there is no RF output.

### Additional context
Add any other context about the problem here.

pmilazzo, 2025-07-12:
3.25.5.1 and 20250707-307d9e2f are identically broken. It appears that the reason that the Rocket M9 running 20250707-307d9e2f mostly works is that the frequency on that node was never set using the new UI. It was upgraded directly from an old custom-built version to 20250707-307d9e2f and the frequency settings were not changed.

aanon4, 2025-07-12:
In the two support data files attached, one if from the release and one is from the nightly. The release contains channel -299 while nightly contains channel 5. The latter looks correct.
Upgrading from 3.25.5.1 (which is broken here) to nightly isnt going to fix the channel - but I'd like to know if you can fix it in the UI at that point? What channels are offered to you? Does it always pick -299?

Note: I have no M9 hardware so I cannot try any of this myself.

aanon4, 2025-07-12:
Could you attach the output from `iwinfo wlan0 freqlist` - thanks.

pmilazzo, 2025-07-12:
```
root@K3PGM-M900b:~# iwinfo wlan0 freqlist
  2.387 GHz (Band: 2.4 GHz, Channel -4) [NO_HT40+, NO_HT40-, NO_80MHZ, NO_160MHZ]
  2.392 GHz (Band: 2.4 GHz, Channel -3) [NO_HT40-, NO_80MHZ, NO_160MHZ]
  2.397 GHz (Band: 2.4 GHz, Channel -2) [NO_HT40-, NO_80MHZ, NO_160MHZ]
  2.402 GHz (Band: 2.4 GHz, Channel -1) [NO_HT40-, NO_80MHZ, NO_160MHZ]
  2.412 GHz (Band: 2.4 GHz, Channel 1) [NO_80MHZ, NO_160MHZ]
  2.417 GHz (Band: 2.4 GHz, Channel 2) [NO_80MHZ, NO_160MHZ]
  2.422 GHz (Band: 2.4 GHz, Channel 3) [NO_80MHZ, NO_160MHZ]
  2.427 GHz (Band: 2.4 GHz, Channel 4) [NO_HT40-, NO_80MHZ, NO_160MHZ]
  2.432 GHz (Band: 2.4 GHz, Channel 5) [NO_80MHZ, NO_160MHZ]
  2.437 GHz (Band: 2.4 GHz, Channel 6) [NO_80MHZ, NO_160MHZ]
  2.442 GHz (Band: 2.4 GHz, Channel 7) [NO_80MHZ, NO_160MHZ]
  2.447 GHz (Band: 2.4 GHz, Channel 8) [NO_80MHZ, NO_160MHZ]
  2.452 GHz (Band: 2.4 GHz, Channel 9) [NO_80MHZ, NO_160MHZ]
  2.457 GHz (Band: 2.4 GHz, Channel 10) [NO_HT40+, NO_80MHZ, NO_160MHZ]
  2.462 GHz (Band: 2.4 GHz, Channel 11) [NO_HT40+, NO_80MHZ, NO_160MHZ]
  2.467 GHz (Band: 2.4 GHz, Channel 12) [NO_HT40+, NO_80MHZ, NO_160MHZ]
  2.472 GHz (Band: 2.4 GHz, Channel 13) [NO_HT40+, NO_80MHZ, NO_160MHZ]
  2.484 GHz (Band: 2.4 GHz, Channel 14) [NO_HT40+, NO_HT40-, NO_80MHZ, NO_160MHZ]
```

aanon4, 2025-07-12:
https://github.com/aredn/aredn/pull/2325

pmilazzo, 2025-07-13:
This fix appears to solve the problem. Nightly build 20250707-307d9e2f works on both a Rocket M9 and a NanoStation loco M9. I tried various combinations of 907, 912, 917, and 922 MHz center frequencies and 5, 10, and 20 MHz bandwidth. In each case I observed the expected signals on my spectrum analyzer. I also tested communications between the two nodes at both 5 and 20 MHz bandwidth.