aredn/aredn #2338: TPLink CP710 problem in Mesh mode (adhoc), with a solution

grozzie2, 2025-07-20:
New here, so excuse me if this is posted incorrectly.

A clear and concise description of what the bug is.

Two TPLink CP710-v2 units, 400 feet apart, consistently fail when in Mesh (adhoc) mode, 10mhz chanel, tested on ch 149 and 180.  They actually fail on 5, 10 and 20mhz wide channels.

Not a beginner with openwrt, and altho I have been building and using it for 20 years, this is my first foray into working with adhoc mode. 

I've spent some time troubleshooting this issue.  Went back to basics and put a bog standard openwrt on both of them, they still fail in the same manner using an  ad-hoc network on the rf path.

Looking at kernel logs, they produce an error message from the ath10k driver to the effect of 'Invalid HT mode'.  Also of interest, that more often than not happens when running an iperf between them, and at the same time that error comes up, there is a segment fault reported, backtrace points to an iperf library, but it's not the source of the issue.  Source of the issue is an errant pointer from the ath10K driver.

Turns out there is an easy solution.  Set NOHT rather than HT20 in the htmode field.  Yes, they run a bit slower, but, they no longer produce driver errors referencing an invalid ht mode.  Iperf tests run start to finish without an error, continuously for over an hour.

I checked out the aerdn repository, built from scratch with no changes, the error does reproduce with my build flashed on.

I then made a one line change to /usr/local/bin/node_setup

In the stanza setting up the radio for mesh mode, I added
htmode="NOHT"

After a rebuild and flash onto the units, they now run flawlessly, over the RF path.

What I am not sure of, not very familiar with this code base, is what I can reference at this point to make this change unit specific, or if it even should be unit specific?

I know some chipsets seem to work with HT20 on the narrow channels, but I also believe that HT20 is indeed invalid for a 10 or 5mhz channel.

FWIW, these units (qca9880 wireless) also fail in adhoc mode on 20mhz channels with HT20 set for radio mode, but they dont fail anymore with NOHT set.

Another point of interest, the aredn build puts the dd-wrt firmware in place of the ath-10k-ct which openwrt uses 'out of the box'.  With this NOHT change in place, these units run fine using the ath10k-ct firmware.

I have some more units here that are not in the supported list, which I am in the process of working out what they can, and cannot do.  At least one of the fails on narrow channels, haven't got to testing a couple of the others yet.  Not sure if folks here want submissions for new hardware support that wont do 5 and 10mhz channels.

aanon4, 2025-07-20:
TPLink doesnt get a lot of love anymore, so this is rather useful.

We know that NOHT vs HT20 makes a difference on 10MHz channels on Mikrotik devices (as one example), so we can't just change this for everything. That means we need to understand when we need NOHT and when not. I dont really want to just force this chance on specific models and would rather detect when it can be used.

Could you start by adding the support data for you device. Thanks.

grozzie2, 2025-07-21:
I suspect this is less about the brand, and more about which chipset.  I mistyped above, the cpe710 has the qca9888.  I've got another access point here with the 9880, I've got everything apparently working on that, except narrow channels, been working on sorting that out.  I also have another one here with the 9888 like the tp-link, just working up a build for that one.  I'll have a lot more info once I have these going correctly.

As for when to set NOHT vs HT20, I think the correct path (my opinion only) is to go with NOHT as the default, then upgrade to HT20 on chipsets where it is known to work, and I also wonder if this is a 'per chipset', or 'per device' issue.  I have two devices here that use the 9888 for 5ghz wireless, one appears to work with 5 and 10mhz channels, and the other not.  I'm just in the process of re-doing the build for those 2 and make sure they have the exact same driver / firmware set for 5ghz.  The other difference, one is a dual radio unit (yuncore xd4200),  If I can get everything working on that one, it's a great unit for this application.  It is housed in an industrial metal outdoor enclosure, with 4 N connectors for the antennas.  It came with 4x 6db omni, but, could be fed into anything.

Another unit that has befuddled me, should be a trivial addition, ubiquiti unifiac-mesh.  Nice little outdoor dual radio unit.  With the stock antennas removed, and the flat panel screwed on, it's a 15db sector setup.  I have more than one of them.

What has befuddled me, it should be trivial, except, the aredn stuff looks for ART data in /dev/mtd7, it's in /dev/mtd8 on this one.  I hacked all the references to mtd7 to look at mtd8, added a correct stanza in radios.json with the correct identifier in all ubiquiti stuff, but that didn't do the trick.  It still comes up saying 'radio unknown' and uci wireless has nothing other than the two phy sections.  It's like the aredn init wiped out the default stuff, then put nothing back in it's place, and I cant find where that happens.  A pointer to that would be immensely helpful.

I'll get that support data shortly, just re-doing a few thing to take a bunch of my debug / troubleshoot stuff out.

grozzie2, 2025-07-21:
Support data from the first one

[supportdata-VA7NM-house-2025-7-20-20-8.tar.gz](https://github.com/user-attachments/files/21339687/supportdata-VA7NM-house-2025-7-20-20-8.tar.gz)

Support data from the second one

[supportdata-VA7NM-dome-2025-7-20-20-8.tar.gz](https://github.com/user-attachments/files/21339690/supportdata-VA7NM-dome-2025-7-20-20-8.tar.gz)

The units are 400 feet apart, set to 5mhz channel 180 when I took this.  I restarted them (power cycle via remote power switch) then I ran multiple iperfs in both directions after they showed connected, before taking the data, to confirm I was in a working configuration.  They were the only two running when I took this data.

The network is set up with both of them in smart switches, there is a 10gb optical fiber path between the remote end and switch and the switches in house here.  The one labelled 'dome' is the far one, it has untagged data on the ethernet port, which is the path my workstation takes to get in.  It's wan is from the switch feeding vlan 1, vlan 2 is disabled at the switch.  The one labelled house has all 3 vlans disabled at the switch, so the only path to it is via the rf link.

aanon4, 2025-07-21:
Thanks. As this is custom firmware let me confirm a few things:

1. You are using NOHT
2. You are using stock firmware (not the dd-wrt firmware)
3. HT20 doesn't work with this firmware on 5, 10. What about 20 bandwidth?

As I recall, tplink is the only hardware we currently support which uses the qca9888 (I need to check this). This keeps the risks associated with any changes here nicely compartmentalized.

grozzie2, 2025-07-21:
I agree, was thinking about this a bit more.

We mention qca9888 with that one because that's the driver we load.  In fact, according to the commit data on openwrt site, the tplink has a qca9896 chipset.  I have another access point here (yuncore xd4200) that does have the qca9888 chipset, and I have confirmed, it does NOT do 5 and 10mhz channels, with either firmware on board, I have yet to try the -ct-full-htt variant which is what gets loaded onto the litebeam.

I am not finished combinations and permutations yet, but, I can say this much fairly definitively, I'm thinking the unit can be stabilized, it just may have to deal with lower performance in adhoc mode.

I have started making up a matrix of what does, and does not work.  The outstanding question in my mind, if the tplink is set to NOHT, will it still work reliably connecting to something like  a rocket set to HT20 on the other end.  On a first rub, against the litebeam it appears to.  

I'll have the rocket here in a couple days, it's the unit destined to go up on the mountain next month for wide area coverage.  That's going to be my ultimate test case, does it work correctly with the rocket feeding it's sector antenna.

In the meantime, been pondering something else.  I noticed in the radios.json there is an 'exclude bandwidths' section in the openwrt one definition.  Maybe that is the correct place to put a stanza in for 'use NOHT', as it does appear to be a device specific thing.  I have looked carefully at the outputs from iw phy, and can find nothing to differentiate reliably there.  The surprising bit I found in that comparison, the tplink suggests it can do 160mhz channels, whereas most of the other ones I have here cannot, and the qca9888 very definitely does not.

aanon4, 2025-07-21:
So I need to clarify/note a few things:

1. QCA9888 is only used by TPLink CPE710 v1 and v2. I assume you've only tested v2.
2. All recent tests are with dd-wrt firmware.
3. NOHT works for 5 & 10. Did you test NOHT with 20 or with VHT20 or HT20?

I'd like to finalize this today if possible. I'll jump thought the PR hoop as I already have branches for this done.

aanon4, 2025-07-22:
https://github.com/aredn/aredn/pull/2341

closed 2025-08-09 by aanon4