aredn/aredn #2343: Mikrotik LHG5ac  fails Traceroute or obtain Internet connection from Neighbor RF node.

k1ky, 2025-07-22:
### Describe the bug
A clear and concise description of what the bug is.
Node fails to fully connect to neighboring rf node. Traceroute "Hangs" on search for IP addys 8.8.8.8 and neighbor node ID addy. No Internet path where other nodes succeed connected to same target.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes, both 3.25.5.1 and Nightly 0718

### Expected behavior
A clear and concise description of what you expected to happen.
Unit shoud be able to connect and navigate to other nodes.

### Support data
Pl

[supportdata-W4WWF-LHG5AC-BWDRPT-12TWR-359-2025-7-21-22-2.tar.gz](https://github.com/user-attachments/files/21357894/supportdata-W4WWF-LHG5AC-BWDRPT-12TWR-359-2025-7-21-22-2.tar.gz)

ease add the support data for all nodes mentioned in this bug report (**Required**)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

k1ky, 2025-07-23:
Neighbor shows connected, but no iperf or traceroute success.

[image]

grozzie2, 2025-07-25:
There is a pull request pending with changes for this unit.  I've got an LHG 5 ac XL here, and with the changes in the pending pull request it runs stable.

FYI, the changes are essentially a band-aid to work around an issue with the ath10k drivers, performance is reduced in Mesh (adhoc wireless) mode, but the unit will run stable